portapack-mayhem/mayhem-firmware #249: Bug Detector

W54J04S07T, 2020-12-10:
This seems like an obvious feature for a device that can scan the entire spectrum in 1-6000Mhz in 1 second.

Wondering if the spectrum analyzer can be tweaked to act as a "bug detector", or more specifically something
that can ID nearby RF emitters of any type?  Potentially the ability to also display signal strength as a means
to identify direction via antenna rotation might also be possible?

Has this been done, or is there a work around that allows this to be done with existing features?

euquiq, 2020-12-10:
I would think that the "Looking Glass" app should cover the basics.

In any case, consider that Hackrf is a bit "deaf" ... specially on high frequencies.

W54J04S07T, 2020-12-12:
"a bit 'deaf'"  it may be, OTOH, I consider it a bit of a wonder that it works at all
without wave guides at those higher frequencies.

Given the needs of signal travking in general, I was however expecting to have
to use directional antenna"s" ( note the plural ) anyway.

I like Looking Glass.  Can't find the git source for it.  Any pointers?

eried, 2020-12-13:
Hi, it is here https://github.com/eried/portapack-mayhem/pull/215

GullCode, 2020-12-14:
You can also take all the ranges from here and put them in a freqman formatted file so the search/the scanner can load it.
http://www.tscm.com/bugfreq.html
and here:
http://www.tscm.com/bugfrqmfr.html
And when you'll have make a freqman file with it, share it with us :-D

W54J04S07T, 2020-12-16:
LOL, your links got my attention in a big way.  
Then for some reason or other my PortaPack decided it could no longer
see my microSD card ( or any other cards "content" or write to it... this is 
going to set me  back a bit...

I'm hoing to post an enhancement for Looking Glass as a result of this.
Se my new entry.

ArjanOnwezen, 2021-05-28:
As mentioned above. Looking Glass should cover the basics.

closed 2021-05-28 by ArjanOnwezen