portapack-mayhem/mayhem-firmware #250: TETRA Support

JBx0, 2020-12-14:
**Is your feature request related to a problem? Please describe.**
I'd love to be able to capture or even listen in on TETRA signals 

**Describe the solution you'd like**
Basically the same features as the lib offers:
telive is a program which can be used to display information like signalling, calls etc from a Tetra network. It is also possible to log the signalling information, listen to the audio in realtime and to record the audio. Playing the audio and recompressing it into ogg is done via external scripts.

 
**Describe alternatives you've considered**
https://github.com/sq5bpf/telive

rbenzing, 2020-12-17:
https://gitlab.com/larryth/tetra-kit might be an alternative solution as well

nemesis567, 2021-05-08:
Is anyone actively working on this, would like to get into it as it's a feature I'm looking into

dreirund, 2024-11-14:
Why was this closed?

I do not see any note that says that it is implemented or ultimately rejected.

The referenced feature request #1199 was closed because it was a duplicate of this one.

So I request re-opening.

Regards!

gullradriel, 2024-11-14:
Because it's already asked in an open issue : https://github.com/portapack-mayhem/mayhem-firmware/issues/360

gullradriel, 2024-11-14:
So, no. We will not reopen this one, @dreirund .