portapack-mayhem/mayhem-firmware #251: SD card access over usb

woofwoof, 2020-12-14:
**Is your feature request related to a problem? Please describe.**
I needed to physically remove the SD card to access it.

**Describe the solution you'd like**
I guess this may have been asked before but is it at all possible to access the SD card from a device over usb please? Today I had to physically remove it to get to a signal. Do-able, of course, yet a little awkward. I have no idea where to start to try to implement this. Any pointers please? Thank you.

**Describe alternatives you've considered**
Physical removal.

**Additional**

eried, 2020-12-14:
This would be an excellent addition

GullCode, 2020-12-14:
I love the idea. I was even thinking of making something radio based to push / pull files, but an access over usb would be even better:-)

MacGnG, 2021-12-31:
Check this out, hopefully it will solve this problem or get us in the right direction. 

"**_USB Mass Storage Class Host + Filesystem Mounter static library for Nintendo Switch homebrew applications._**"  ......    [https://github.com/DarkMatterCore/libusbhsfs](https://github.com/DarkMatterCore/libusbhsfs)

The UMS (USB Mass Storage) libraries provide this functionality to the Nintendo Switch for hombrew applications. It may be possible that this or another UMS library could provide the functionality requested for the HackRF+PortaPack. 

**What is UMS(USB Mass Storage)**
[https://www.samsung.com/levant/support/mobile-devices/what-is-ums-usb-mass-storage/](https://www.samsung.com/levant/support/mobile-devices/what-is-ums-usb-mass-storage/)

``UMS is a set of computing communications protocols.
Defined by the USB Implementers Forum that run on the Universal Serial Bus. The standard provides an interface to a variety of storage devices.
Most current mainstream operating systems include support for USB mass storage devices, although support on older systems is available through patches.``

Basically how this works in the GUI is: Open a specific menu. Click the SD CARD button and the SD card will show up on the computer as a drive. After the task is completed on the computer, safely eject the drive. The menu displays the device has ejected. and continue as normal.

TriniHam, 2022-03-28:
> **Is your feature request related to a problem? Please describe.** I needed to physically remove the SD card to access it.
> 
> **Describe the solution you'd like** I guess this may have been asked before but is it at all possible to access the SD card from a device over usb please? Today I had to physically remove it to get to a signal. Do-able, of course, yet a little awkward. I have no idea where to start to try to implement this. Any pointers please? Thank you.
> 
> **Describe alternatives you've considered** Physical removal.
> 
> **Additional**
> 
> There is a **[$40 open bounty](https://www.bountysource.com/issues/95190173-sd-card-access-over-usb?utm_campaign=plugin&utm_content=tracker%2F182166508&utm_medium=issues&utm_source=github)** on this issue. Add to the bounty at [Bountysource](https://www.bountysource.com/?utm_campaign=plugin&utm_content=tracker%2F182166508&utm_medium=issues&utm_source=github).

Hey did you make head way with this issue?

elixx, 2022-08-19:
I have to remove the top of the portapack to get the SD card out because of bad slot alignment and a shoddy slot spring. This would be awesome.

Added to bounty.

eried, 2022-08-19:
Yeah, sometimes is difficult. I have some shim on the slot to "align" the sdcard

DienoX, 2023-01-14:
Please, if someone could implement this idea.

gullradriel, 2023-01-14:
Dunno maybe a load moar money would attract a Dev ?
X+D

DienoX, 2023-01-14:
So now I'm wondering if the solution below would be better:
https://aliexpress.com/item/4000188292311.html
https://hardwarerecs.stackexchange.com/questions/13256/seeking-microsd-memory-with-bluetooth-or-wifi
Plus an SD to microSD adapter.
A very cheap solution, but its price is not the most important thing.
Using it, you can conveniently send items to portapack not only from large computers but also from a smartphone.

DienoX, 2023-01-14:
Conveniently I mean without any added cables or physical connections.

euquiq, 2023-01-14:
hmm @DienoX I would suspect that the radio emission from such a card would "ruin" whatever you might want to do with your Portapack ...

smannchen, 2023-01-14:
https://www.ebay.com/itm/402662448009?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=z9mmdtalqj6&sssrc=2349624&ssuid=uf0vw8szt0k&var=672937876085&widget_ver=artemis&media=COPY

DienoX, 2023-01-14:
@euquiq you mean interference other than wifi? Wi-Fi is very close to us but not as close as a microSD card with wifi outside the partapack metal case.

eried, 2023-01-16:
For my 3d printer I bought a commercial wifi sd card that makes a server. At the end I never used it because it is tricky to modify the filesystem while things are running. Besides the interference I think also there is little usability to gain. The sd card slot in the PP sucks, in this case I think a longer microsd would be a good solution while we dont have access over usb :/

DienoX, 2023-01-16:
Certainly, the microSD extension cable seems to be the cheapest and easiest way to connect and disconnect the microSD from the portapack. In mine there is such a problem with the socket in the housing is poorly fitted. To remove the card you need a screwdriver or a tool similar to tweezers.

eried, 2023-01-16:
> To remove the card you need a screwdriver or a tool similar to tweezers.

Yeah, this is the main issue why this functionality is needed. In the case I designed, I tried hard to make it accesible, and I think is better than on most of the metallic cases, but still not good enough.

bernd-herzog, 2023-04-01:
As requested this feature is now implemented.

https://github.com/eried/portapack-mayhem/wiki/SD-Over-USB

closed 2023-04-06 by bernd-herzog

woofwoof, 2023-05-17:
So sorry but which version is this implemented in please? My most stable version doesn't have these menu options. Thanks.

gullradriel, 2023-05-17:
In Mayhem latest stable, or latest nightly, in 'utilities' menu

woofwoof, 2023-05-17:
Thank you. I'll double check. All best.