portapack-mayhem/mayhem-firmware #252: Security access with directional push buttons pattern code.

artur-com, 2020-12-15:
Would be more safer having restricted access to portapack.

In some cases it can be dangerous for the portapack owner if other persons can simply press one button and see the entire menu and thus wrongly suspect of crime activity (or rightly). Non-technical persons go crazy seeing HACK word on the screen and googling MAYHEM.
I have got portapack H2+ where switching on/off is with pressing thumb wheel only and don't feel comfortable when anyone else can have access with simple one press of this biggest button.
A menu option displaying (after pressing on/off button) a custom picture with waiting for push buttons correct sequence before going to main menu would be much appreciated.

reclaudio, 2020-12-16:
Login and password access may be ? 

A: eried/portapack-mayhem
Cc: Subscribed
Oggetto: [eried/portapack-mayhem] Security access with directional push buttons pattern code. (#252)

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

artur-com, 2020-12-16:
Login to where?
Simple custom key sequence on splash screen to pop up main menu is fair enough.

> Login and password access may be ?

eried, 2020-12-16:
There was a play-dead mode on havoc which I removed because it did not work, but it was a similar concept (simulate that the device was not working), however I am not sure about the whole premise of security here. Do you have a lock on your radios and transmitters? 

I am not against it either, just think it is a gimmick.

artur-com, 2020-12-16:
What lock you mean?
I travel a lot by airplane and I am really afraid of security luggage check if they just wanted to turn on the device and have seen revealed main menu with all "hack" stuff at airport. Also at work I don't want to self explain co-workers if they played portapack when was left unattended.

Custom picture splash screen with defined key sequence to start main menu should be enough. I would upload a picture of a DAB+ radio logo or similar to splash screen and set key sequence to avoid start menu with just simple one click.

> Do you have a lock on your radios and transmitters?

h0us3c4ll, 2024-07-08:
@bernd-herzog : thanks for flashing , you mention , a "disarmed" version in #855  which one is that ?

bernd-herzog, 2024-07-08:
@h0us3c4ll there is none. Just the possibility to make one.