portapack-mayhem/mayhem-firmware #253: Looking Glass features

W54J04S07T, 2020-12-16:
Can Looking Glass be enhanced to record a file of
the spectrum it surveys?

Can it be adjusted to record a file of the surveyed spectrum 
that is above or below a stated signal strength threshold?
Can the signal strength threshold be "windowed" ( i.e. above a
certain threshold and below another )?

And finally the real kicker... can Looking Glass be enhanced to
create a file of a surveyed spectrum that can then be used as
an IGNORE file?  In effect, while it conducts a new survey with
an IGNORE file in place, it only records new signals not
found in the IGNORE file???

Would make a h___ of a survey meter now wouldn't it?

euquiq, 2021-01-06:
The "window" feature is easy. The other feature may be impossible, I think, I may be wrong but several things are against it: 

1) The wider the bandwidth, the more amount of data being read (and there is no microSD ram buffer, so no reliable way of storing such amount of data).

2) The sampling being done is not "perfect" (i.e. the listening routine jumps over the "blind spots" at the center and edges of each frequency slice, so the data you would get recorded would be kinda of sketchy / imprecise.

I think the Looking Glass tool is good for visual representation of the selected bandwidth range, but nothing else. 

EVEN the actual filtering (filtering OFF / MID / HIGH) is actually challenging, against the overall app speed.

I could try to add to that filter, more options lke ONLY LOW, ONLY MID, ONLY LOW AND HIGH ... or even try fit in some numeric fields for user-input of one or two windows. But ANYTHING you do, will impact -more or less- on scanning speed

W54J04S07T, 2021-01-20:
Sounds like an invitation for a portapack "H3+++" maybe?  LOL

While the hackrf can scan 1-6000Mhz in under a second, that doesn't preclude 
a MUCH More leisurely spectrum survey.  Breaking up the entire spectrum into
manageable chunks ( what ever size that might be ) and "taking its time" to progress
thru the entire hackrf bandwidth would alleviate a lot of H/W limitations.

Slower, of course has its defects, but nobody is expecting mil-spec combat
ready performance from a fairly cheap "toy" for ham enthusiasts.

AndKe, 2022-02-05:
is it me, or does it lack an option to look at a narrower frequency range?
I would, for example, fid it useful to be able to scan 425 ...445Mhz  ,    but only huge chuncks like 240..500 seems to be possible. (?)

euquiq, 2022-02-05:
For scanning up to 20Mhz you can use the RECEIVERS->Analog Audio app in Spectrum view

AndKe, 2022-02-06:
Thank you.
Is there a reason there is no feely definable range option in the Looking Glass?

gullradriel, 2023-05-10:
Parts of the demands have been solved with recent updates.
The other requests does not seem doable.

closed 2023-05-10 by gullradriel