portapack-mayhem/mayhem-firmware #255: Receive SSTV

gadjodilo83, 2020-12-18:
Is it possible to receive and decode an sstv signal direct with the portapack? 

The International Space Station (ISS) will be transmitting Slow Scan TV (SSTV) in late December to celebrate 20 years of amateur radio operations... :)

[image]

eried, 2020-12-18:
@jamesshao8 was working on this (he posted on discord) but for the moment there is nothing

mad-developer-donald, 2021-05-24:
Hi,
Something new about this functionality? 
Donald here

JoshuaMorris, 2021-05-26:
I would really like to use this functionality also.

eried, 2021-05-26:
There is no further updates on this, but as I said before @jamesshao8 was working on his fork for his custom portapack, so a big bounty here will probably convince him to finish the implementation :)

teixeluis, 2021-05-26:
That would be cool. How far into the implementation have @jamesshao8 went?

teixeluis, 2021-05-26:
Would the portapack have a cam, and it would be all about SSTV selfies.

hotaru2k3, 2021-05-26:
> Would the portapack have a cam, and it would be all about SSTV selfies.

sending SSTV is already supported, this issue is about receiving.

teixeluis, 2021-05-26:
> > Would the portapack have a cam, and it would be all about SSTV selfies.
> 
> sending SSTV is already supported, this issue is about receiving.

Yes I've already tested it. It works quite well. Sent using the portapack, and received with MMSSTV for example.

[image]

JoshuaMorris, 2021-05-26:
Ideally, It would be nice to both send and receive in the same app. My use case being for sending and receiving images for Amateur radio.

jamesshao8, 2021-12-28:
> That would be cool. How far into the implementation have @jamesshao8 went?

Oh, I just implemented sending and receiving for robot8BW on PortaPack. In future, I would probably do NOAA receiving, which is similar but more interesting.

teixeluis, 2021-12-28:
> > That would be cool. How far into the implementation have @jamesshao8 went?
> 
> Oh, I just implemented sending and receiving for robot8BW on PortaPack. In future, I would probably do NOAA receiving, which is similar but more interesting.

Nice one @jamesshao8. For ISS we will need PD-120, but it's great nevertheless. Looking forward for more on this.

Thanks

Cheers

ea3iav, 2022-07-22:
Any advance? I don't even remember how much I put on the bounty 🤔

closed 2023-12-04 by eried