portapack-mayhem/mayhem-firmware #259: Can portapack decode GPS signal?GPS Receiver?

gfdfdxc, 2020-12-26:
Since there's a 0.5PPM TCXO in portapack, can portapack decode GPS signal and show the current location on the map?
Or even use GPS to sync the time?

eried, 2021-02-01:
I wonder the same. It would be cool. I am gonna post a very small bounty on this to see if there are interested people on at least researching it.

Dirk275, 2021-02-02:
On the Portapack H1 board you'll find a place for an Ublox GPS receaver. With this hardware update and with a fitting firmware it would be possible to receive and decode GPS.
Without this modification HackRF One and Portapack will not be able to get the GPS information.
Reasons:
1. Even with an active  GPS antenna it will not be easy to simply receive the native 1.542 GHz signal(s).
2. The interpretation of the GPS data needs more than one receiver to spot all satellites in reach.
3. The receiver channels have to be very accurate and sensitive (all identical) and with exact timing in a nanosecond range.
4. The protocoll must be decoded in real time.

That exactly is the task of the highly specialized GPS receivers (like Ublox 8 series).
It would be very hard to build this similarly with normal electronic parts and Portapack doesn't contain this functioning.

So:
No, the HackRF with Portapack will never be able to decode the GPS.
But "we" could solder the missing GPS parts on the H1 board and then update the firmware to use all GPS information, like done here: https://github.com/eried/portapack-mayhem/issues/230

jamesshao8, 2021-12-28:
> On the Portapack H1 board you'll find a place for an Ublox GPS receiver. With this hardware update and with a fitting firmware it would be possible to receive and decode GPS. Without this modification HackRF One and Portapack will not be able to get the GPS information. Reasons:
> 
> 1. Even with an active  GPS antenna it will not be easy to simply receive the native 1.542 GHz signal(s).
> 2. The interpretation of the GPS data needs more than one receiver to spot all satellites in reach.
> 3. The receiver channels have to be very accurate and sensitive (all identical) and with exact timing in a nanosecond range.
> 4. The protocoll must be decoded in real time.
> 
> That exactly is the task of the highly specialized GPS receivers (like Ublox 8 series). It would be very hard to build this similarly with normal electronic parts and Portapack doesn't contain this functioning.
> 
> So: No, the HackRF with Portapack will never be able to decode the GPS. But "we" could solder the missing GPS parts on the H1 board and then update the firmware to use all GPS information, like done here: #230 ... with the other missing parts.

1. Receiving and Decoding GPS signal through HackRF is possible, but the processing has to be done on a computer, it's not possible realize on a PortaPack.
video: https://www.bilibili.com/video/BV1Mp4y147dc
2. Receiving and Decoding GPS through UBlox chip is also possible, and with that PortaPack can show position in realtime. I already soldered them on H3.
video: https://www.bilibili.com/video/BV1Xq4y157Sr

ArjanOnwezen, 2022-04-21:
Option isn't feasible without adding GPS chip.

closed 2022-04-21 by ArjanOnwezen