portapack-mayhem/mayhem-firmware #262: Settings not saved after reboot

ShutdownRepo, 2021-01-05:
**Description / steps to reproduce**
When using mayhem on a H2 portapack, I try to save settings like UI (splash, backlight timeout), time (date and hour) but a reboot sometimes erase that config. Time is reset at every reboot.

**Expected behavior**
The user config to not reset when rebooting the device

**Affected versions**
Latest (1.3.0) affected, not tested on previous versions

ShutdownRepo, 2021-01-05:
Thank you for your quick answer, I didn't think about the rtcc, I'll check this asap and make sure this is the cause of the issue

NiteCrwlr, 2021-01-06:
This is not a bug, just put in a battery and make sure the internal lipo is not creating a short-circuit.
I think you need a CR2025.

eried, 2021-01-07:
The H2 most certainly needs a CR1225 
https://github.com/eried/portapack-mayhem/wiki/Hardware-overview#portapack-internals

closed 2021-01-07 by eried

ShutdownRepo, 2021-01-09:
Thank you for you help 😊 the issue (that indeed was not one) is now resolved