portapack-mayhem/mayhem-firmware #264: Audio/SDcard issue, then solved: WM8731 support

Xelibix, 2021-01-08:
I recently acquired a portapack h2 model here: https://aliexpress.com/item/4000907775687.html
Then I flashed the latest 1.3.0 mayhem firmware.
I wasn't able to use the headphone jack and I couldn't boot when the SD card was plugged in the portapack.
I found I had to flash a modified firmware from here : http://www.mmdvm.club/index.php/h2.html.
Now everything is working.
The modified firmware if probably here to support another version of the audio chip wich is the wm8731 intead of the ak4951 on other h2 revisions.
It would be a good Idea to have a support for that new type of board.

eried, 2021-01-08:
Oh but this is quite bad that they changed the chipset 🙄 I don't see the source code for the new one. I would suggest returning or avoiding those units

euquiq, 2021-01-12:
You will depend on those guys for having an updated firmware "forever" and you will never know exactly what are they doing inside the firmware. Maybe it ends up being sub-par with the open-sourced version, due to inconsistencies, or just bad implementation!

cabottech, 2021-01-20:
Just a heads up for others who may stumble on this issue thread: I think I have a similar issue with a PortaPack H2 purchased here: https://banggood.app.link/3EoKBoaTbdb

I also can't boot with an SD card installed, but once booted I can insert and use the SD card.

If I go to Debug > Peripherals > AK4951 I get a page of zeros.

I was unable to download the alternative firmware (linked by OP) :(
I'll tinker with this a bit and see if the seller can provide more details.

cabottech, 2021-01-20:
Just a heads up for others who may stumble on this issue thread: I think I have a similar problem with a PortaPack H2 purchased here: https://banggood.app.link/3EoKBoaTbdb

I also can't boot with an SD card installed, but once booted I can insert and use the SD card.

If I go to Debug > Peripherals > AK4951 I get a page of zeros.

I was unable to download the alternative firmware (linked by OP) :(
I'll tinker with this a bit and see if the seller can provide more details.

eried, 2021-01-20:
Hi, I would try to ask for a refund and warn other users via negative feedback until they actually make a piece of code that works for all portapacks. In my opinion, that custom version .bin wont be maintained for long and users will have a slighty different unit which cannot even use the original portapack fork

cabottech, 2021-01-20:
I've requested a return/refund, so we'll see.

Here's a photo of the device I received, confirming it has the WM8731 chip. Hopefully this helps others to identify the **incompatible** hardware. 
[image]

Can someone please post a photo of good compatible hardware please?  It may even be worth putting this in the Wiki (I noticed the warning added to the README).

eried, 2021-01-20:
It is not a perfectly clear picture, but it looks like this: https://github.com/eried/portapack-mayhem/wiki/Hardware-overview#portapack-internals

It would be interesting to see how your refund will go because that is not a non-functional/broken unit. In fact, if the manufacturer made a clean code patch that worked for all units it would be a perfectly fine alternative, but in this current state: people receiving those units will stop getting any support after their "buyer protection" runs out on the ecommerce site.

Worse is that some of these units are being shipped with a current version of this repo, but with the patches already applied, so they are not going to notice any problem for weeks.

Xelibix, 2021-01-20:
Hi cabottech use one of this links to dowload the "alternative" firmware that support the WM8731 audio chip:
link 1:  http://www.taozhengbo.com/mayhem_v1.3.0_FIRMWARE_8731.zip
link 2 : https://www.mmdvm.club/usr/uploads/2020/12/2458645348.zip

cabottech, 2021-01-20:
> Worse is that some of these units are being shipped with a current version of this repo, but with the patches already applied, so they are not going to notice any problem for weeks.

@eried That's very frustrating. I didn't get any custom firmware or any instructions with my device, so I'm glad to have caught this early. I've just noticed the product page shows the correct (original) hardware with the open hardware logo and all. So I can argue that I have received an incorrect device which may help my return/refund...

> Hi cabottech use one of this links to download the "alternative" firmware that support the WM8731 audio chip:
> link 1: http://www.taozhengbo.com/mayhem_v1.3.0_FIRMWARE_8731.zip
> link 2 : https://www.mmdvm.club/usr/uploads/2020/12/2458645348.zip

@Xelibix Thanks for that. I'll download a copy in case I get stuck with this device.

eried, 2021-01-20:
> So I can argue that I have received an incorrect device, which may help my return/refund...

Yes, if it goes thru it will be very helpful so all other people receiving those units can do the same and ask for refunds as early is possible.

MasterXuuu, 2021-01-25:
seems here :http://www.mmdvm.club/index.php/archives/139/ 
show how to support wm8731

W54J04S07T, 2021-01-26:
[quote] Oh but this is quite bad that they changed the chipset 🙄 I don't see the source code for the new one.[/quote]
@eried
These guys are Chinese. They did publish the patches, but... finding it ( in Chinese )
may be a bit more challenging going forward.  The H/W is actually not bad.

eried, 2021-01-26:
Sure, they probably have reasons. But until they do not submit the changes here, but include a precompiled version of this fork in their hardware, the recommendation is still **to avoid all those units, and ask for a refund the earliest possible.**

In 3 months all the users with those units will want to upgrade to a new release, and they will discover that it is not possible and the issues will be opened here, not in chinese forums.

closed 2021-01-26 by eried

Stein3247, 2021-12-04:
i have the same chip WM8731 on my portapack h2.
and i have almost the same issue.
i bought it on aliexpress, got it delivered with v 1.4.0 and everything was working fine until 
i tried¨to upgrade firmware from 1.4.0 to 1.42. 
now the unit wont boot up. i have tried to follow your link to the modified firmware, but of course all of the link is broken.
so do any of you guys have the modified firmware on your pc somewhere or could modify firmware 1.4.2 for me ?
because i dont have any clue about how to modify it.
best regards stein

Brumi-2021, 2021-12-04:
Sorry , but I am confused,  in our  Mayhem FW code ,  I think that have already sw code files  detecting  and soporting both audio  IC's no?   AK4951 and WM8731 , not ? Or are talking  about different special  WM8731 IC version?  
[image]

sck-nogas, 2021-12-04:
Duplicating this comment here too...

Hmmm, if you have a WM8731, then [the commented out code on line 184 in "firmware/application/portapack.cpp" is the issue](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L184)..

I created a kludge of a [pull request](https://github.com/eried/portapack-mayhem/pull/439) to let us build a version of the firmware for the older audio chip, since it currently can't be detected in code. 

Or you can edit your code and comment out the "[model = PortaPackModel::R2_20170522](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L187)" line and uncomment the "[model = PortaPackModel::R1_20150901](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L184)" line.