portapack-mayhem/mayhem-firmware #265: Problem upgrading the firmware 1.2 to 1.3 mayhem

iz2fly, 2021-01-12:
1- HackRF One Portapack H2 (Chinese December 2020)
2- Firmware 1.2
3- Windows 10
4- First update
5- HackRf One is in Device Manager
6- HackRF One works with SdrConsole and other programs
7- Impossible to update
[image]
[image]
[image]

Thank you all.

eried, 2021-01-12:
Hi, have you tried in DFU mode?

iz2fly, 2021-01-12:
Yes.
[image]

iz2fly, 2021-01-12:
PROBLEM SOLVED now with driver update!
I was using the Zadig driver.
Why complicate simple things?

Now let's see what can be done with HackRF One,
maybe it will be possible to TX in SSB in the future?

eried, 2021-01-12:
What you did to fix the issue? Can you describe? You mean the .inf included in the compressed release file?

iz2fly, 2021-01-12:
1- In HackRF One mode
2- Win 10, go to Device Manager
2- Find HackRF One
3- "Update Driver"
4- "Search for drivers on your PC"
5- Go to the DRIVER folder (uncompressed) inside the "mayhem_v1.3.0_FIRMWARE" folder
6- Update driver
7- Reset button, done.

[image]

iz2fly, 2021-01-12:
Then obviously firmware updated with default procedure.

iz2fly, 2021-01-12:
[image]

eried, 2021-01-12:
Ah nice! that is described in the video with the instructions :) but it is handy to have in case someone else encounters the same problem.

closed 2021-01-12 by eried