portapack-mayhem/mayhem-firmware #266: APRS: TX, but no RX?

stl1988, 2021-01-14:
(Still have to get my Portapack, so this is based on what I have seen in videos and images)
Is there a reason for APRS being only available in the Transmit sectiom, but not in the Receive section? It doesn't really make sense to me.

GullCode, 2021-01-14:
It's AFSK when receiving

stl1988, 2021-01-16:
I have got my Portapack now. But when trying AFSK decoding on 144.8 MHz (Europe) I only get garbage, even when there's an APRS signal.

GullCode, 2021-01-17:
App is in yellow in the menu, which mean it's not fully implemented. If you're feeling you can help, feel free to have a look in the code :-)

euquiq, 2021-01-18:
It is my understanding that there is NO APRS receive. it is NOT AFSK when receiving. It is NOT done, directly. Sorry to bear "bad news".

Actually APRS TX app was full of bugs, as it came from Havoc. I did fix it (it took me several hours, and a complete "revamping" of the code because the error was at packet level, wrongly formed).

jamesshao8, 2021-01-19:
https://www.youtube.com/watch?v=yQ9aqo682B0

euquiq, 2021-01-19:
Nice, James Shao! if you could share that code, it would be truly appreciated mainly by the HAM radio guys. I fixed the APRS TX algorithm some months ago, and have been trying to purchase a radio with aprx TX so to be able to decode the messages and clone / modify the afsk app so to have an aprs rx, but you already did it. Kudos!

ArjanOnwezen, 2021-05-03:
> https://www.youtube.com/watch?v=yQ9aqo682B0

Hi @jamesshao8 , it would be cool indeed if you could share the code / do a PR.

East2West, 2021-05-03:
@ArjanOnwezen I implemented my own version of APRS RX in the latest release (1.4.0)

https://github.com/eried/portapack-mayhem/pull/325 

This issue can probably be closed now since it's a duplicate of https://github.com/eried/portapack-mayhem/issues/158

ArjanOnwezen, 2021-05-03:
Great work @East2West

closed 2021-05-03 by eried