portapack-mayhem/mayhem-firmware #268: CLKIN leaks 10Mhz reference frequency

berkinkartal, 2021-01-20:
Dear All,

As explained by RFNoob in https://github.com/eried/portapack-mayhem/issues/79 CLKIN port of Hackrf with Mayhem firmware emits/leaks 10Mhz reference signal. Moreover, I observed some unexpected behavior during some tests with gnuradio in this respect:
CLKIN port leaks HackRF internal reference clk during normal operation and doesn't allow an external reference clock to be used (according to specs that should be a CMOS square wave clock, DC coupled). I tried this also with various firmware versions but no luck. Has anyone ever tested this?
Also I tried setting up a simple Signal generator using gnuradio. When using HackRF mode with portapack (v1.3.0), there's a little sinus clock sign on portapack screen, it always shows EXT (guess it refers to an external clock), although HackRF generates configured signal properly, eventhough without an external clock source connected.
Another observation is that if you connect this CLKIN port to a frequency counter or a scope with 50ohm termination or just connect a 50ohm terminator to it, clock source sign on portapack LCD turns to HRF (I suspect that means, hackrf internal clock).
And I couldn't manage to feed an external clock into hackrf so far.

Expected behavior:
Hackrf CLKIN port doesn't emit a 10Mhz signal in any situation. And you observe the right CLK source (internal) in Mayhem firmware menu (options->radio), "Reference Source". Because currently what I'm seeing there is only "External 10.0000MHz".

euquiq, 2021-01-20:
@berkinkartal I just plugged my OSC into CLKOUT, I turn my CLKOUT ON and see the appropirate clk on my oscilloscope screen. I turn it off, and everything goes quiet at my oscilloscope. Then I move my oscilloscope probe into the CLKIN, and all I can see is a quiet signal, no leak at all. I put my Portapack in scanner mode, or REC mode, Or even turn on and off again the CLKOUT, and the CLKIN keeps quiet.

berkinkartal, 2021-01-20:
Hi,
@euquiq I connected CLKOUT port of my Hackrf with a 50ohm terminator to the scope, set Hackrf output frequency to 10Mhz from Portapack and observed the result in "scope_8.png" (1V/div, 1.33V peak-to-peak).
Then without changing anything connected the same cable to to CLKIN port and observed the output in "scope_9.png" (200mV/div, 200mV peak-to-peak). Settings of the scope you can also find in the screenshots.
Enabling or disabling CLKOUT has no effect on the behavior of CLKIN, also power-cycling hackrf has not effect either.
I have a clone, on the PCB of HackRF it says "13 February 2014". I reckon that's an old one.
Not sure if have a faulty one though...

[image]
[image]

berkinkartal, 2021-01-20:
@euquiq Another observation, I removed Portapack daughterboard entirely, powered Harckrf from USB and connected the scope input to CLKIN, now I see nothing on scope output. 
Is there a tool to test CLKOUT from Hackrf tools?

berkinkartal, 2021-01-20:
I just tested CLKIN ref input without Portapack and I'm pretty sure it works without it (I started thinking the problem is with Portapack, may be hardware). Connected CLKIN to a square wave source with CMOS output as mentioned in the documentation, created a simple signal source configuration from gnuradio and started the executed the setup. Observed the output signal from scope, modified the source and and observed from scope, did several tests to see it's working properly or not. Then I disconnected CLKIN source port during the time gnuradio config was running, generated signal from HackRF was gone.

euquiq, 2021-01-21:
It seems you have something bad going on on your hardware. Maybe it's the portapack itself, given that hackrf's clkin starts behaving ok when you test it unplugged from your Portapack.

Which portapack is it ? H2 ? maybe it is one of the latest ones, which include a different sound IC ? can you show some pictures of it so maybe we can better identify it ?

Are you still inside the "return" window for the portapack / hackrf purchase ? maybe you can claim a warranty issue ?

euquiq, 2021-01-28:
Shall we close this issue ?

pa0o, 2021-01-28:
Tried to use external 10MHz ref but without success.
clkin shows 1V tt 10.000 MHz
clkout shows 1.6Vtt 14.586 MHz
so no clue how this can work.
Portapack with its TCXO is very stable and accurate at freq.
So ist not a big deal

Another thing is that the external ant 3V2 is not present in Radiosonde mode
 where a preamp is needed for fox hunting grounded transponders

jirjirjir, 2021-04-12:
> @euquiq Another observation, I removed Portapack daughterboard entirely, powered Harckrf from USB and connected the scope input to CLKIN, now I see nothing on scope output.
> Is there a tool to test CLKOUT from Hackrf tools?

@berkinkartal 
I think we must divide these two issues: one with leaking of 10M from CLKIN and other with correct CLKOUT output.

Here I'll try to setting out my opinion of probable cause of first one.
Looking at schematic of Portapck and HackRF we can see that only way to get 10M freq out from CLKIN is Portapack's REF_CLK signal connected to CLKIN pin 2 of P22 inter-board connector which routed to HackRF's CLKIN of Si5351 (also to CLKIN P16 SMA connector where you attached your scope).

According code investigation there are folowing steps produced by HackRF's firmware to determinate which clock source to use during its boot up process:
1) check for external source on CLKIN of HackRF's Si5351 by inspecting of bit 4 of its register 0 (sets for 1 if 5351 got correct signal), if it's exist stop following checks and continue boot up;
2) try to wake up Portapack's TCXO and re-check for external source on CLKIN of HackRF's Si5351, if it's exist stop following checks and continue boot up;
3) start using Si5351's local oscillator.

In the case I suppose that you are falling to step 2 bypassing (for some reason) step 1. 

Could you plaese clarify, in which exact moment you attached your external reference source to CLKIN connector: _before_ powering up the system or during its operation?

gullradriel, 2023-04-26:
Closing due to inactivity / not reproducible

closed 2023-04-26 by gullradriel