portapack-mayhem/mayhem-firmware #269: Transmission SSB mode

iz2fly, 2021-01-20:
I recently bought HackRF One with Portapack h2, I am very excited because I see many possible applications.
Most important to me would be to be able to use SSB in TX mode.
I've been waiting for a radio like this for a long time.
I saw a video of test made in 2014, I don't know if anyone has already done it for Portapack h2. Who can do this? 
Please let me know.

https://www.youtube.com/watch?v=cjyE59tvbbc

texasyojimbo, 2021-01-22:
This is the same as #84.

ultraohm, 2021-03-11:
SSB RX/TX this option would be spectacular if implemented

iz2fly, 2021-03-12:
> SSB RX/TX this option would be spectacular if implemented

Already done. We are testing how it works. Look:

https://github.com/eried/portapack-mayhem/issues/198

ultraohm, 2021-03-12:
Great News thx Ernesto, i will test too when arrive my portapack, 73´s bye
CT2KCK op Paulo

 

On Sex Mar 12 10:27 , iz2fly  sent:

SSB RX/TX this option would be spectacular if implemented

Already done. We are testing how it works. Look:
#198

—
[
{
"target": "https://github.com/eried/portapack-mayhem/issues/269#issuecomment-797394872",
"url": "https://github.com/eried/portapack-mayhem/issues/269#issuecomment-797394872",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

edwardstorlien, 2021-04-29:
I vote for this too! Would be awesome with an HF amplifier

teixeluis, 2021-05-31:
I guess this issue can be closed, even though the current implementation seems to have issues, at least with some types of hardware.

closed 2021-05-31 by eried