portapack-mayhem/mayhem-firmware #270: transmit with flex pager protocol

craphole, 2021-01-23:
The ability to transmit with the flex pager protocol

eried, 2021-02-01:
Do you have equipment to test all that? a flex pager? You could open a bounty on this, send a pager to somebody who wants to work on your bounty. Otherwise is very unlikely that somebody would work on this.

patvdleer, 2021-02-03:
RX would be nice given the P2000 system in the Netherlands runs on this, I can read the message via 

`rtl_fm -f 169.65M -M fm -s 22050 -p 83 -g 30 | multimon-ng -a FLEX -t raw /dev/stdin`

craphole, 2021-02-18:
I have an rtl-sdr with sdrsharp software to recieve the signal and pdw pager decoding software to decode the flex pager signal.

epvuc, 2021-02-22:
Flex receive (and sure, TX also) would be pretty great, as it's getting to be much more prevalent than POCSAG. multimon-ng can decode Flex as well as POCSAG with no problem, so it's probably not terribly different.

ellisgl, 2024-10-06:
Flex TX is pain compared to POCSAG. Constant broadcasting and slots have to be in the correct place and you can only send a page 1 per four minutes to a pager iirc