portapack-mayhem/mayhem-firmware #299: Sandisk 16 GB SDCard not working

njbrown09, 2021-01-27:
For some reason, it appears that i cannot get my sdcard to load

Here is the image of the sdcard
https://gyazo.com/54a9da3a4a094aa6f45aa235621e69b5

I looked around and someone said in another issue it might be a chinese knockoff version, but i tried their firmware and it didnt work either,

eried, 2021-01-27:
Can you get any sdcard to work?

njbrown09, 2021-01-27:
I have a weird offbrand one that makes the indicator green but writing fails with the test.

njbrown09, 2021-01-27:
The only other issue i have with this is that the volume on headphones and the built in speaker are very quiet

eried, 2021-01-27:
You can use the knob to raise the volume in apps that play audio. It is one of the values that go from 0 to 100

For the sdcard, you need to get other samples. Also do not rely on that test, load the files and check if you can open for example the GPS samples in the GPS replay app. If that works, you are good to go.

njbrown09, 2021-01-27:
That was my theory, but it says "no files" and file manager says empty filesystem.

eried, 2021-01-28:
And you did format it in fat32 and copy the files? can you retrieve all the files back in a computer?

njbrown09, 2021-01-28:
Yes

eried, 2021-01-28:
Then try another sd, if does not work the portapack is faulty

njbrown09, 2021-01-28:
Damn, okay

eried, 2021-01-28:
Check the soldering pads around the sd slot, sometimes just touching them with a soldering iron will fix the problem if one pin looks funky

eried, 2021-01-28:
I only use https://www.banggood.com/Mixza-Year-of-the-Dog-Limited-Edition-U1-32GB-TF-Memory-Card-p-1295486.html 😂 (really, for all my gopros and devices)

njbrown09, 2021-01-28:
Hmm okay. My knockoff sd card mounts and shows a green status, but the device says its only 127mb and fails to write to it.

njbrown09, 2021-01-28:
https://gyazo.com/06e07991c2d38d931ec37fcef54dad61

eried, 2021-02-01:
Format it in FAT32 4K cluster size, load the files and try again. If you can load the map, and take captures, then it is fine, do not use the debug options to check it.

closed 2021-02-16 by eried

gilmarwsr, 2021-09-15:
If, like me, you are using a MacOS try to format the SD card as Master Boot Record. Worked for me.

Commander-Crash, 2021-09-25:
To anyone this may help:

I had this issue thinking it was the PortaPack H2 when in reality it was the hackrf. I found this out when I ordered a PortaPack to replace the issue. It did the same with the new Portapack so I had a busted Hackrf One lying around that had its antenna ripped off. I flashed that one and wallah it worked. So before digging around in the sd card slot with a butcher iron check your hackrf one 1st.

AndreiSoroka, 2021-11-04:
[image]
I had a problem:
`Fail: File Open Write`

CD card 32gb

**My solution:**
1. Record [mayhem_vX.Y.Z_COPY_TO_SDCARD.7z](https://github.com/eried/portapack-mayhem/wiki/Update-firmware#microsd-card-files) to sd card.
2. Tools > Wipe SD card
3. Debug > SD Card

**How not work for me:**
- Clear SD card (Fat 32/Fat 32 + Wipe/Ntfs + Wipe)
- Photos and other trash (Fat 32/Fat 32 + Wipe)
- mayhem_vX.Y.Z_COPY_TO_SDCARD.7z on the card (Fat 32, btw, got another error)

[image]

And problems: After wipe MacOs can't read SD card... Need to format again

eried, 2021-11-05:
@AndreiSoroka but at the end, everything works perfectly?

AndreiSoroka, 2021-11-05:
@eried yep
[image]

But! You will have a problem with SD Card on the computer after Wipe. (I checked on MacOs, and in Discord checked on Linux)

And need to format fat32 again

[image]

Brumi-2021, 2021-11-07:
In my case, after trying Mayhem **Wipe SD** , the Mayhem Debug SD test was working well (and I could see my SD speed, latency parameters (very good) , but as a side effect, after Wipe tool,  I couldn’t read any more that SD from my laptop. (As my previous above comment , “failed to mount 16 Gb volume”)
So far, the  way that I found from Linux  , to recover the micro SD card (from that strange state) without formatting it , was , first find the name of that disk-unit with : **sudo fdisk -l**,   (in my case  : /dev/sdb1  .     Then  send that  "Check FAT filesystem on DEVICE for errors" Linux command : sudo **fsck.vfat -f -p /dev/sdb1**,   and that is it !  Now again the SD card is back to the original state before using Wipe SD . That means , ok in PC, working well in Mayhem Capture recording and replay  and all,  except the Debug test , that keeps Failing  : compare file.  So still not solved all , but at least found the way to recover it to previous state before Wipe tool without formatting SD. But looking for final solution help  , (that means, to have , no errors in Debug  SD ,nor in the laptop, no problem in capture-replay) . 
[image]

dominch, 2023-01-03:
My old 8G card died and I needed get new one for portapack. The one I've got was 32GB and it should still be ok. 
I formatted it as default FAT32, copied files and inserted into portapack - altou it display OK on sd card icon - card was not recognised. I went to utils and choose wipe sd card (to create right filesystem) and then inserted it to pc card reader. This time card was not recognised by system and I was unable to copy anything there.
I found out somewhere that it should be FAT32 with 4k block size, tried to create such partition but it was not possible for such large partition (default block size is larger), so I just created 16GB partition, formatted it as FAT32 with 4k block size, copied files and then it showed correctly on portapack. 

Of course half of card is unused now, but I don't need it that much. Maybe in future firmware will be able to talk with bigger cards, right now easiest is to created smaller partition.

frednice, 2024-02-18:
I did almost the same and it worked for me!
I had a 64GB card which works now in 8GB / FAT32 / 4K block size.
When I formatted it to 16GB, Windows then could not read the card, that's why I decreased to 8GB
However, the SD card testing fails
[image]

apple601601601, 2024-04-13:
I had a similar problem with a 32GB sandisk SD card, but didn't solve it after trying the following:
* create a 32G partition and format with FAT32 and default block size
* create a 16G partition and format with FAT32 and 4k block size

I fianlly solved the problem by changing the partition table format from GPT to MBR.(Somehow the card was formatted to use GPT before).

It doesn't matter what block size I use after changing to MBR, both 4K/default block size works, and I was able to use full space of the SD card.

Hope it helps.

gullradriel, 2024-04-13:
That's good information.

hamid321321, 2025-02-05:
Download the version of the firmware file and connect the memory card to the computer and install the RAM file with the application on the RAM and the RAM will be completely fixed.