portapack-mayhem/mayhem-firmware #301: Portapack doesn't boot with microsd inserted

ckuethe, 2021-01-28:
**Describe the bug**
Portapack doesn't boot with microsd inserted

**To Reproduce**
Steps to reproduce the behavior:
1. Insert MicroSD
2. Power on Portapack
3. Observe black screen
4. wait a few minutes
5. observe continued black screen

or

1. Power on Portapack
2. Observe successful boot, with missing sdcard icon
3. Insert MicroSD, observe sdcard ok icon
4. press the hackrf reset button
5. Observe black screen
6. wait a few minutes
7. observe continued black screen

**Expected behavior**
I expect (but I could be wrong) that the portapack would boot with or without a microsd card inserted.

**Affected versions**
1.3.0
1.3.1

Just got a brand new hackrf+portapack
haven't tried bisecting old firmwares
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

**Additional**
Samsung 32GB EVO Plus microsd card.  No I/O errors reported by dd when trying to fill or dump the card contents, comparing checksums between the contents of the card and the contents of the sdcard zip file show no content differences so I feel like the card isn't fake or damaged.

euquiq, 2021-01-28:
You either have a faulty Portapack or microSD. If you tested your microSD in other device and works fine (Format it in FAT32!!) but it crashes your Portapack, then you may -just "in case" try a different microSD on your Portapack (preferably 32gb or less capacity one). 

If problem persists, and you  just purchased your Portapack then you may need to ask for a refund. 

In any case, this is not an issue we could do something about in here.

ckuethe, 2021-01-28:
OK, thanks. I'll reformat it again to be sure, though when I insert the card after it boots the datafiles are all useful (eg. the map, airlines, frequency lists)

eried, 2021-01-28:
This is not something documented before, but I agree with @euquiq

ckuethe, 2021-01-28:
```
# fdisk -l /dev/sdb
Disk /dev/sdb: 29.81 GiB, 32010928128 bytes, 62521344 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xba7ebb84

Device     Boot Start      End  Sectors  Size Id Type
/dev/sdb1          32 62521343 62521312 29.8G  b W95 FAT32

# mkfs.vfat -n PORTAPACK -F 32 -v /dev/sdb1 
mkfs.fat 4.1 (2017-01-24)
/dev/sdb1 has 64 heads and 32 sectors per track,
hidden sectors 0x0020;
logical sector size is 512,
using 0xf8 media descriptor, with 62521312 sectors;
drive number 0x80;
filesystem has 2 32-bit FATs and 32 sectors per cluster.
FAT size is 15264 sectors, and provides 1952836 clusters.
There are 32 reserved sectors.
Volume ID is f47f0820, volume label PORTAPACK  .
```

For completeness I tried all combinations of: with and without the DOS compatibility flag`; partition types `0xb` (FAT32) and `0xc` (FAT32 LBA); with and without a volume name/label.

Thanks for confirming that Mayhem should boot with a microsd present.

closed 2021-01-28 by ckuethe

cchangeur, 2021-04-17:
Same problem here just after an update (1.3.0 to 1.3.1). Problem present after downgrade :( 
So, faulty Portapack is the only acceptable response ?

ckuethe, 2021-04-18:
I just reflashed with the latest nightly next (`hackrf_spiflash -w ~/hackrf/portapack_nightly_202104180016_next.bin `) and my portapack now boots with an sdcard inserted.

ckuethe, 2021-04-22:
Portapack now also boots with Mayhem 1.4.0

Stein3247, 2021-12-04:
i have the same issue, bought my portapack h2 with hack rf from aliexpress.com.
it arrived with mayhem v1.4.0
and it was working perfectly, until i was downloading the new firmware from github, 
and installed V1.4.2
after that i just have a black screen, i have tried finding the modified firmware from link 1: http://www.taozhengbo.com/mayhem_v1.3.0_FIRMWARE_8731.zip
link 2 : https://www.mmdvm.club/usr/uploads/2020/12/2458645348.zip
but all of the chinese sites are down. could anybody help me getting my portapack up and running.
maybe one of you guys have downloaded the modified firmware 1.4.0 or v1.4.2 and could share the firmware here ? 
best regards stein

eried, 2021-12-06:
You can try older versions from here: https://github.com/eried/portapack-mayhem/releases