portapack-mayhem/mayhem-firmware #303: Morse delay

vdlame, 2021-01-31:
hello, I tested the transmission of Morse with an application which decrypts the audio its works great except the first letter is always false, it is the transmission delay which makes its, a pause of 0.5 second would be perfect, and  for more flexibility if we could choose the time it would be great.  otherwise you are doing a great job, I can't wait to see future versions

eried, 2021-01-31:
What happens if you send several E followed by an S, how many you lose?

vdlame, 2021-02-17:
its works by starting with an E followed by a space

Example:e hello
And the app Morse code agent decode and receive : hello
It work !!

gullradriel, 2023-04-28:
Since we found a workaround but the bug isn't solved, I mentioned it in the wiki: https://github.com/eried/portapack-mayhem/wiki/Morse