portapack-mayhem/mayhem-firmware #304: Portapack-H2 update firmware, headset has no sound

miaoyuwelcome, 2021-02-01:
Hello, after Portapack-H2 is updated to mayhem_v1.3.1_FIRMWARE or mayhem_v1.3.0_FIRMWARE firmware, run flash_portapack_mayhem.bat, restart the device and find that the headset has no sound. Turning on or off the speaker icon in the status bar at the top of the screen does not work, is it incompatible?

closed 2021-02-01 by miaoyuwelcome

eried, 2021-02-01:
Did you solve this?

miaoyuwelcome, 2021-02-02:
> Did you solve this?

Found that the H2 module is a third party! The audio decoding is WM8731, the firmware is not compatible, the firmware must be specified, which is very bad

eried, 2021-02-02:
Ahhh, yes :/ That is why I added the warning because that is going to generate a lot of issues which are silent now.

eried, 2021-02-02:
I would suggest to ask for a refund to convince the manufacturers to follow the right way to make those changes (i.e. do PR that does not break the current users devices, but also support the new WM chip)

miaoyuwelcome, 2021-02-04:
> I would suggest to ask for a refund to convince the manufacturers to follow the right way to make those changes (i.e. do PR that does not break the current users devices, but also support the new WM chip)

I have applied for return, I will support genuine Portapack-H2