portapack-mayhem/mayhem-firmware #305: Aircraft description from ICAO database

UserAZERTY0, 2021-02-05:
Hi,

Would it be possible to give aircraft information from an ICAO database in the ADSB module?
There are some quite complete ones available on the Internet, for instance: https://opensky-network.org/datasets/metadata/
From my pov I would find this more interesting than the call sign / airline / country (especially some aircraft won't give such info like police or military...) - or at least it could be given on top of these information when available.

KR,
Max

eried, 2021-02-05:
Can you give a small example with a draft for example of the suggested new info? It would help to understand the request (i.e. grab a photo of the current UI and just write on top to open a discussion here)

UserAZERTY0, 2021-02-06:
Well, ideally a second button with “aircraft info” next to “see on map” would be great :-)
As there are 5 to 10 fields of interesting information generally mentioned.

Here is an example I just recorded: aircraft with ICAO “396673” was seen by Portapack but nothing else (no callsign no gps intel…)
Luckily it is in the database: its registration is "F-GZTT", aircraft model is "Boeing 737-48E (SF)", operator is "Asl Airlines France S.a. - France" and its engines are "CFMI CFM56-3C1".

If to be displayed on a dedicated page (through a 2nd button), the following fields would be nice :
“icao24”, "registration", "manufacturername", "model", "serialnumber", "operator", "operatorcallsign", "owner", "engines".
Otherwise maybe putting all the fields in the dedicated page would be preferred - so that if we find a more complete database, code will be built once for all?

On the current detailed page, adding "manufacturername", "model" and "owner" would still be very cool !
I created a proposition based on a screenshot as you asked.

Not all the fields seem to be present in the database found (and some ICAO were not found through some aircrafts I have tried, but still with +460K entries there is a nice likelihood aircrafts will be found).

Note: within the database, ICAOs do not seem to be ordered, to ease the “finding process” (and its speed), I guess it might be necessary to organize it and use a kind of a recursive search algorithm…

KR,
Max
[image]
[image]
[image]

GullCode, 2021-02-06:
Should be doable. What is the size and format of the database to use ? Best would be something like a csv text file

UserAZERTY0, 2021-02-06:
It is the case indeed: a 80MB csv file with +460K aircrafts: https://opensky-network.org/datasets/metadata/aircraftDatabase-2021-01.csv
Seems to be the most comprehensive free maintained database.

GullCode, 2021-02-06:
Looks like it is interesting informations indeed.
if I understand correctly, once we have an ICAO then if someone is clicking on 'Aircraft informations' we just have to read line by line until we find a matching ICAO and then display the available fields on screen.

UserAZERTY0, 2021-02-06:
Yes it is basically that, with if possible information displayed on 2 levels:

1) On the existing detailed page, with 3 main fields displayed like "manufacturername", "model" and "owner" + recalling the ICAO somewhere else perhaps to gain space (in the title next to Details ?)
2) A 2nd button leading to a dedicated page with all the fields displayed
Personally I would put these main information on top (detailed page), because before being from company A or country B, an aircraft is firstly identified by its “technical” description (ICAO, model, owner - owner might indicate the company btw).

Please note that there is a possibility ICAO is not found in the database (Opensky claims there is a 85% match - which is pretty good).

Regarding the database, something more efficient than line by line search should be possible, like ordering the csv file by ICAO alphabetical order (Excel would do it), then a "recursive search like" should be quicker (but it would require to load the whole file - not sure Portapack is able to do that...) or maybe an index could be built to accelerate the search (if needed).

UserAZERTY0, 2021-02-06:
Yes it is basically that, with if possible information displayed on 2 levels:

1) On the existing detailed page, with 3 main fields displayed like "manufacturername", "model" and "owner" + recalling the ICAO somewhere else perhaps to gain space (in the title next to Details ?)
Personally I would put these main information on top (detailed page), because before being from company A or country B, an aircraft is firstly identified by its “technical” description (ICAO, model, owner - owner might indicate the company btw).

2) A 2nd button leading to a dedicated page with all the fields displayed

Please note that there is a possibility ICAO is not found in the database (Opensky claims that there is a 85% match - which is pretty good).

Regarding the database, something more efficient than line by line search should be possible, like ordering the csv file by ICAO alphabetical order (Excel would do it), then a "recursive search like" should be quicker (but it would require to load the whole file - not sure Portapack is able to do that...) or maybe an index could be built to accelerate the search (if needed).

UserAZERTY0, 2021-02-06:
Yes it is basically that, with if possible information displayed on 2 levels:

1) On the existing detailed page, with 3 main fields displayed like "manufacturername", "model" and "owner" + recalling the ICAO somewhere else perhaps to gain space (in the title next to Details ?)
Personally I would put these main information on top (detailed page), because before being from company A or country B, an aircraft is firstly identified by its “technical” description (ICAO, manufacturer, model, owner - owner might indicate the company btw).

2) A 2nd button leading to a dedicated page with all the fields displayed

Please note that there is a possibility ICAO is not found in the database (Opensky claims that there is a 85% match - which is pretty good).

Regarding the database, something more efficient than line by line search should be possible, like ordering the csv file by ICAO alphabetical order (Excel would do it), then a "recursive search like" should be quicker (but it would require to load the whole file - not sure Portapack is able to do that...) or maybe an index could be built to accelerate the search (if needed).

GullCode, 2021-02-06:
Load the whole file => not possible

UserAZERTY0, 2021-02-06:
Without building an index, then the simplest way would be:
- to order the csv file by ICAO alphabetical order (Excel would do it)
- to do a line by line search until you find your ICAO or until you know you missed it (because searched_ICAO > current_ICAO = not present)
Hence at least you won't run the whole file for nothing if your ICAO is not present.

A second simplest option would be to run the file backwards if searched_ICAO > half_file_ICAO and a third simplest way would be to build an index with "ICAO range" vs "database entry points" (so not a complete index).

UserAZERTY0, 2021-02-06:
Without building an index, then the simplest way would be:
- to order the csv file by ICAO alphabetical order (Excel would do it)
- to do a line by line search until you find your ICAO or until you know you missed it (because searched_ICAO > current_ICAO)
Hence at least you won't run the whole file for nothing if your ICAO is not present.

A second simplest option would be to run the file backwards if searched_ICAO > half_file_ICAO and a third simplest way would be to build an index with "ICAO range" vs "database entry points" (so not a complete index).

euquiq, 2021-02-06:
I would give a try to just move over the text file, line by line, until finding the correct code. Even if it is "messy", probably the processing time will be more than tolerable. 

It would be nice to get that information on the ADSB app: But then it will be another file to get into the microSD (and mantain fresh) but it sounds "doable"

ArjanOnwezen, 2021-11-28:
@euquiq / @GullCode I can give it a shot, using an index and creating a python script to easily keep it up to date. Just like the recently released airlines.db. Not sure if it is performant enough, but let's see.

ArjanOnwezen, 2021-12-01:
It's a bit slow with a standard index, ala airlines.db. Takes a few seconds. I'll add a binary search tree this weekend.

ArjanOnwezen, 2021-12-10:
ICAO info is available on the current next branch and in the next release.

closed 2021-12-10 by ArjanOnwezen

UserAZERTY0, 2021-12-12:
Thanks a lot ArjanOnwezen !!