portapack-mayhem/mayhem-firmware #306: Portapack H2+ speaker off

IZ4PNN, 2021-02-09:
Hi,
I bought this Portapack H2+ :

https://a.aliexpress.com/_vVoHBV

This looks much better than the previous version, but it has two problems:

1) Can't turn it off while charging (or I don't know how to do it).

2) I have installed an 8Ohm 3W speaker .. the audio is sufficient but every time I have to click the speaker icon to activate it. After a restart, for example, it is inactive again.

eried, 2021-02-09:
For the first part, maybe it is not possible? have you tried double click the central button?

The second, you need a battery installed on the device

IZ4PNN, 2021-02-09:
Hi Eried and thanks for your job.
Yes, I also tried the double click but it doesn't turn off. Others have confirmed that it does not go out.

I have the battery installed, the other parameters and the time are correctly stored. For this reason I have reported this as a bug on H2 +

73

eried, 2021-02-09:
That is very strange, but on a next version the speaker should be disabled as default. Is your device keeping the time?

IZ4PNN, 2021-02-09:
Yes, it seems to correctly hold the time, the clock output (on or off) and the speaker icon (visible or hidden).
The audio comes out correctly in the headphones, in the speaker only if the speaker icon is active but as I told you the speaker icon does not remain activated.

ArjanOnwezen, 2021-05-18:
@IZ4PNN  Do you mean the speaker icon is grey (not green) every time you switch it on?
I have the same thing with my PortaPack H1. It remembers the icon setting, but not the sound (speaker, or in the H1 case the headphone output) setting.
I'll have a look at this this weekend, since it's annoying it needs to be switched  on every time.

teixeluis, 2021-05-18:
Same behavior on my side as well.

IZ4PNN, 2021-05-19:
@ArjanOnwezen yes exactly ... we hope for a fix.

@teixeluis thanks for the info, so does the H2 version that was off while charging have a different chip?

73

teixeluis, 2021-05-19:
@IZ4PNN I only have a device with this particular chip. The device is always on when connected via USB (even if it was powered off initially).

IZ4PNN, 2021-05-19:
@teixeluis same here ....

ArjanOnwezen, 2021-05-20:
I've  build something yesterday. I'll push it to tonight. Please note I only tested this on a portapack H1 with a AK chip. There basically the speaker works as the headphone output.

IZ4PNN, 2021-05-20:
@ArjanOnwezen thank you!
I will test it, I am using an H2+ with AK chip

ArjanOnwezen, 2021-05-20:
Cool then I'll also post a link to the compiled firmware. So it will save 20 minutes compiling ;)

IZ4PNN, 2021-05-20:
Thanks :)

Mirko Dalmonte (IZ4PNN)
MySondy GO Developers Team

If you like our work, please buy us a beer! You can do it directly on our
official page.

https://mysondy.altervista.org

teixeluis, 2021-05-20:
Cool, I will also give that a try once I have the replacement speaker opamp arrived (NS8002 chip) and installed. 

I have accidentally fried the original one after installing a speaker and not noticing that one of the speaker wires was crushed against the chassis when closing it. The bridged audio opamps are not any bit forgiving in respect to such mistakes/accidents, because of their floating signal outputs.

IZ4PNN, 2021-05-20:
@ArjanOnwezen sorry, there is the signature of my project, by mistake 😬

ArjanOnwezen, 2021-05-20:
No problem, but good to see your also contributing to the world ;)

mad-developer-donald, 2021-05-24:
Yes it's not possible to switch off while charging. I confirm this. 
But you can use the Sleep mode (Moon). Without the charging usb cable, you can switch it off with a double click the central button. I have a battery installed,

The speaker symbol is by default off(grey), but the speaker is still working. After I activate the speaker(green) and deactivate again(grey), the speaker is off.

ArjanOnwezen, 2021-05-24:
@mad-developer-donald 
That issue should be resolved by https://github.com/eried/portapack-mayhem/pull/364 where speaker state is saved and not switched off at reboot. Also it will not automatically be switched on during when audio the audio receiver, since state is kept.
@IZ4PNN could you test it with your H2 device.. ?

IZ4PNN, 2021-05-25:
@ArjanOnwezen sorry, not yet ... as soon as I get home I will try 😬

ArjanOnwezen, 2021-05-26:
Great! Just put your remarks below. :)

ArjanOnwezen, 2021-05-31:
@IZ4PNN  Any chance you have time to test sonewhere this week?

IZ4PNN, 2021-06-01:
@ArjanOnwezen I believe today ... I'll update you ... sorry for the delay

IZ4PNN, 2021-06-01:
@ArjanOnwezen tested!
Works as expected.
Only note: the volume is not memorized so every time it is necessary to adjust the volume.
Not a very serious problem :)

Thanks!!!!!

teixeluis, 2021-06-01:
Is there battery backed NVRAM which could be used to keep the state of these controls (speaker mute, volume, current frequency, etc), without adding wear to the internal flash?

And by battery I mean the cell battery, as the LiPo is fully disconnected from the HackRF on power off.

ArjanOnwezen, 2021-06-01:
@IZ4PNN  than

> Is there battery backed NVRAM which could be used to keep the state of these controls (speaker mute, volume, current frequency, etc), without adding wear to the internal flash?
> 
> And by battery I mean the cell battery, as the LiPo is fully disconnected from the HackRF on power off.

Thanks for testing.
@eried  Could merge this after you've tested this one as well

eried, 2021-06-02:
Yes! I will set an alarm to check the PRs tonight 😁

ArjanOnwezen, 2021-06-17:
> Is there battery backed NVRAM which could be used to keep the state of these controls (speaker mute, volume, current frequency, etc), without adding wear to the internal flash?
> 
> And by battery I mean the cell battery, as the LiPo is fully disconnected from the HackRF on power off.

A bit late.. but yes there's persistent storage..

teixeluis, 2021-06-17:
> A bit late.. but yes there's persistent storage..

Hehe yes, the code gave me the response meanwhile.

jaessy77, 2021-06-24:
Hi there,
https://github.com/eried/portapack-mayhem/issues/306#issuecomment-845283734
worked for me .. but the tagged version in the Repo with V1.4.0 not. I binary compared the files, they differ.
Best regards
Marco
(Portapack H2+ from https://www.banggood.com/PortaPack-H2+-with-Shell-Assembled-for-HackRF-One-SDR-Software-Defined-Radio-1MHz-6GHz-0_5ppm-TCXO-with-3_2inch-Touch-LCD-Display-p-1811611.html?cur_warehouse=CN&rmmds=buy)

ArjanOnwezen, 2021-06-25:
True, since this commit hasn't been merged yet.

ArjanOnwezen, 2021-09-28:
@eried  Could PR above be merged?

AndKe, 2022-02-06:
What's the status on this?
Mine too came with the H2+
I remember to have hears the speaker working at some point.
Then I upgraded from 1.4.2 to 1.4.3
The 3.5mm jack works, but not the speaker.  - and I see no way to "switch it on" 
- it is connected internally to the PCB, and the speaker er 8.3ohm.  Tested the pins with an oscilloscope, and there is nothing there.

eried, 2022-02-07:
@AndKe have you tried with the options for showing the "speaker" option? in settings

@ArjanOnwezen I totally missed this comment, is it still relevant? :O

AndKe, 2022-02-07:
@eried so that's where it went, it must have been disabled by default. Thank you - it works as expected now.

PHATLXIV, 2022-03-15:
> That is very strange, but on a next version the speaker should be disabled as default. Is your device keeping the time?

My HackRF One + Portapack H2+ is not keeping time or saving settings. Do you happen to have a solution for this or could you point me in the right direction? I haven't had a chance to take a closer look, but I have a suspicion that maybe the socket isn't properly connected to the board, as replacing the CR1220 battery did not resolve the issue as I had expected. I may be wrong about this whole thing, though. Thank you in advance.

eried, 2022-03-15:
Take a look on the solder joints, it sounds as you suspect, a bad connection

jaessy77, 2022-03-15:
> Hi there, [#306 (comment)](https://github.com/eried/portapack-mayhem/issues/306#issuecomment-845283734) worked for me .. but the tagged version in the Repo with V1.4.0 not. I binary compared the files, they differ. Best regards Marco (Portapack H2+ from https://www.banggood.com/PortaPack-H2+-with-Shell-Assembled-for-HackRF-One-SDR-Software-Defined-Radio-1MHz-6GHz-0_5ppm-TCXO-with-3_2inch-Touch-LCD-Display-p-1811611.html?cur_warehouse=CN&rmmds=buy)

Hi there :-)
Just flashed my device with latest Mayhem firmware v1.4.3 and all is working brilliant.
Only problems I got, probably introduced with Win11 was USB cable in concatenation with recognize the device as HackRf. 
But by trying few cables (now I have a 1m textil based one from www.action.de ) the DFU device LPC was recognized, I could update via DFU .. and voila .. HackRF was seen in Win11. Then usual way with Zadig and all was fine.
Speaker is working, 3,5 mm is working ... Really satisfied  :-)
And I agree to eried, seems the solder points are potentially not that good as those could be. Just take a Pb based solder, heat up and resolder. 
Should solve your problem.
Best regards
Marco

reclaudio, 2022-03-16:
unsubscribe

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

closed 2023-03-31 by gullradriel