portapack-mayhem/mayhem-firmware #307: Looking Glass PRESETS.TXT crashes app

frinkemon, 2021-02-09:
**Describe the bug**

Loading the attached PRESETS.TXT crashes LooingGlass

**To Reproduce**
Steps to reproduce the behavior:
1. Given the attached PRESETS.TXT
2. Open Looking Glass App
3. App does not open, Unit hangs, clock stops.

**Expected behavior**
.. it works.. :)

**Affected versions**
Latest version

**Additional**
File checked for oddities, all txt labels are <=20 chars, 
[PRESETS.TXT](https://github.com/eried/portapack-mayhem/files/5953808/PRESETS.TXT)

euquiq, 2021-02-09:
I would try a shorter list. I suspect you are asking more than what the little ram inside your hackRF can handle.

frinkemon, 2021-02-09:
> I would try a shorter list. I suspect you are asking more than what the little ram inside your hackRF can handle.

Yeah I thought that, but it's only an 8K file, when a new line from the file is added to the vector with presets_db.push_back then if there is a memory allocation issue then the allocator should throw an exception, I'm not sure what happens though.

It'd be good if it could either just load as much as it can or read it in chunks from the file if that wouldn't work, or perhaps implement a file chooser so you can have a few presets instead of just the one. Actually that'd be quite useful.

When I get a dev env up for this I will have to do that :-)

euquiq, 2021-02-10:
"the allocator should throw an exception" ... I may be wrong, but we do not have such luxury. Also, loading 8kb of text and frequencies (int64_t) values,  is absolutely out of the question. (Think about halving it.) 

Maybe a custom user control, which would show "a line from a file" on the selector, and read the file, until reaching the right text line and fetching it, every time you change the selection, may be doable.  Then we would need to modify the app so it can use that single line, "parsing it on the fly", instead an index to an array. It is probably doable.

darknebular, 2022-07-18:
I checked that if you open ADSB receiver, the PRESETS.TXT will be creaded and loaded with other data...

The codification of the .txt file is mandatory, In 1.5.4 with incorrect codification you will be able to hang up this app.

darknebular, 2022-07-18:
Try with UTF-8 Windows CR LF

Best regards.

closed 2023-03-31 by gullradriel