portapack-mayhem/mayhem-firmware #310: Problem building the firmware

ImDroided, 2021-02-13:
Ok guys I just updated to ubuntu 20.04 and I am having issues getting this to work. here is my terminal output below. all the required components are installed already pip and yaml are there. 

twitch@kl0p:~/SDR/hackrf/portapack-mayhem$ sudo docker build -t portapackccache -f dockerfile-nogit .
Sending build context to Docker daemon  446.5MB
Step 1/13 : FROM ubuntu:xenial
 ---> 8185511cd5ad
Step 2/13 : ENV ARMBINURL=https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux.tar.bz2?revision=108bd959-44bd-4619-9c19-26187abf5225&la=en&hash=E788CE92E5DFD64B2A8C246BBA91A249CB8E2D2D 	PATH=$HOME/bin:$PATH:/opt/build/armbin/bin
 ---> Using cache
 ---> 34185943a32d
Step 3/13 : VOLUME /havoc
 ---> Using cache
 ---> 8c67477ebf13
Step 4/13 : WORKDIR /havoc/firmware
 ---> Using cache
 ---> 89c24501ccad
Step 5/13 : RUN apt-get update && 	apt-get install -y git tar wget dfu-util cmake python3 ccache bzip2 curl && 	apt-get -qy autoremove
 ---> Using cache
 ---> b330777e1ab5
Step 6/13 : RUN curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py && 	python3 get-pip.py
 ---> Running in 50ebae125b02
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1884k  100 1884k    0     0  3766k      0 --:--:-- --:--:-- --:--:-- 3776k
Traceback (most recent call last):
  File "get-pip.py", line 24244, in <module>
    main()
  File "get-pip.py", line 199, in main
    bootstrap(tmpdir=tmpdir)
  File "get-pip.py", line 82, in bootstrap
    from pip._internal.cli.main import main as pip_entry_point
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 954, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 896, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1139, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1115, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1096, in _legacy_get_spec
  File "<frozen importlib._bootstrap>", line 444, in spec_from_loader
  File "<frozen importlib._bootstrap_external>", line 533, in spec_from_file_location
  File "/tmp/tmpnmnb1_es/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
The command '/bin/sh -c curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py && 	python3 get-pip.py' returned a non-zero code: 1
twitch@kl0p:~/SDR/hackrf/portapack-mayhem$

ImDroided, 2021-02-13:
If I remove the pip install line from the docker file then I get this output saying yaml not found. this has to be a python versioning issue. sometimes python sucks. 

witch@kl0p:~/SDR/hackrf/portapack-mayhem$ sudo docker build -t portapackccache -f dockerfile-nogit .
Sending build context to Docker daemon  446.5MB
Step 1/12 : FROM ubuntu:xenial
 ---> 8185511cd5ad
Step 2/12 : ENV ARMBINURL=https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux.tar.bz2?revision=108bd959-44bd-4619-9c19-26187abf5225&la=en&hash=E788CE92E5DFD64B2A8C246BBA91A249CB8E2D2D 	PATH=$HOME/bin:$PATH:/opt/build/armbin/bin
 ---> Using cache
 ---> 34185943a32d
Step 3/12 : VOLUME /havoc
 ---> Using cache
 ---> 8c67477ebf13
Step 4/12 : WORKDIR /havoc/firmware
 ---> Using cache
 ---> 89c24501ccad
Step 5/12 : RUN apt-get update && 	apt-get install -y git tar wget dfu-util cmake python3 ccache bzip2 curl && 	apt-get -qy autoremove
 ---> Using cache
 ---> b330777e1ab5
Step 6/12 : RUN pip install pyyaml
 ---> Running in f0246ccb3dd8
/bin/sh: 1: pip: not found
The command '/bin/sh -c pip install pyyaml' returned a non-zero code: 127

ImDroided, 2021-02-14:
Hey everyone I have had a hell of a time building this on a fresh install of ubuntu. I have tracked the issue to the deprecation of python 2.7 as it reached its end of life on 01/01/2020. Anyone building with the docker script on a newer system possibly even old will fail the pip install part and everything after. The fix for me was to force a 2.7 pip install by changing the docker script download link.
#Install current pip from PyPa
#RUN curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py && \
RUN curl https://bootstrap.pypa.io/2.7/get-pip.py -o get-pip.py && \
python3 get-pip.py
It seems like we need to figure out how to make the move to python3 as they are dropping support for python 2

closed 2021-02-14 by ImDroided

eried, 2021-02-14:
Thanks for posting the solution, maybe it should be added to the wiki if this is a known problem