portapack-mayhem/mayhem-firmware #314: enhance the wiki page for developing app interface and behaviour

marcoSanti, 2021-02-15:
Hi everyone, I was starting to try to code my app, but i found that there is almost no guide on how to create the interface or how to  code the behaviour of my app. Since I am new to c++ (but no to OOP), I am finding myself in a very hard situation. could someone help me out? 
At this point I have managed to add my empty app, but now (even if I look around to other apps code) I have no idea where to start.
To be honest i think that, the lack of documentation is what is stopping more people to start improving this great project...
Thanks in advance!
P.S. someone should be improving in particular this page https://github.com/eried/portapack-mayhem/wiki/Create-a-simple-app
PPS Someone should create a list of available API / syscall in order to make the development easier... I would do it myself but I barely understand where files are located right now...

eried, 2021-02-16:
Yes, we know. Sadly mostly of the code is undocumented in a friendly way. If you go to discord you could ask people who already made some simple mods and apps but we are all just learning. 

Ideally, that page will get completed and be friendly in the future, but as you see in the [latest commits](https://github.com/eried/portapack-mayhem/commits/next), most of the effort now has been into fixing small bugs and cleanup.

eried, 2021-02-16:
I forgot to add, that the wiki is completely open to contribution, so if you dig into making your first app, please complete that initial guide to help future "marcoSanti"s 💯

GullCode, 2021-02-18:
He did it :-)

marcoSanti, 2021-02-18:
well, more than did it, i started doing it :)

eried, 2021-02-18:
Yes, I saw! Amazing, keep it up 😇

ArjanOnwezen, 2021-11-28:
Duplicate of #46

closed 2021-11-28 by ArjanOnwezen