portapack-mayhem/mayhem-firmware #315: Compiling stopped...

loewal, 2021-02-15:
Hi to all developers:

I noticed that it's some weeks ago that compilations could be downloaded.
Is there some special reason for this?

Grtz. Loe

jwetzell, 2021-02-15:
Are you talking about the nightly builds?

loewal, 2021-02-15:
Yes indeed.

eried, 2021-02-16:
Hi, check: https://github.com/eried/portapack-mayhem#where-is-the-latest-firmware 

There is a link for the latest, and another link for the nightly builds. The naming of those is generally: "name_datetime-branchname.bin"

loewal, 2021-02-16:
Thanks Erwin,

That information was already known to me…
I was just thinking why there is no update anymore since January 18….?
At that time every 3 days there was a new build (but perhaps no changes…)?

Grtz. Loe

eried, 2021-02-16:
Ahh, I see what you mean!

The service that compiles the nightly bin seems to have an error, I will try to fix it today.

[image]

eried, 2021-02-16:
It is fixed now.

closed 2021-02-16 by eried

loewal, 2021-02-17:
Great!