portapack-mayhem/mayhem-firmware #316: Spectrum drawing

UserAZERTY0, 2021-02-18:
Hi,

Just found this fun idea: https://www.gkbrk.com/2021/02/spectrum-drawing/
Giving the following outcome - for the ones not willing to read the whole post: https://gkbrkmedia.b-cdn.net/articles/2021-spectrum-video/btc.mp4

As Gokberk uses HackRF functions, we might be able to "easily" integrate this drawing feature to the Portapack.
The application interface could be reused from the existing SSTV one (sending a picture).
Eventually, beyond sending pictures, we could even think about a Portapack Pictochat based on spectrum drawing :-)

Of course reading/viewing can be done through another Portapack (Looking Glass) or any SDR hardware with common apps (SDR#...)

KR,
Max

eried, 2021-02-19:
That is cool 😂 it would be nice if you can actually also write text directly

bernd-herzog, 2023-05-20:
Closed as this is now available

closed 2023-05-20 by bernd-herzog