portapack-mayhem/mayhem-firmware #318: Flex pager Protocoll support

Micha-123, 2021-03-01:
is it possible to add also FLEX protocoll there is only POCSAG on the mayhem firmware. but in NL the live P2000 network use FLEX protocoll. is it possible to add it to the mayhem firmware?

https://en.wikipedia.org/wiki/FLEX_(protocol)

P2000 as used in The Netherlands
Frequency: 169,650 MHz
Speed: 1600 bit/s (2FSK)
Protocol: Motorola FLEX

epvuc, 2021-03-16:
I've put a bounty on this protocol support.  It's more common than POCSAG in the US as well, as far as I can see. What I'd like to see from the bounty is the abilty to receive and decode FLEX like [multimon-ng](https://github.com/EliasOenal/multimon-ng) ideally for both 2FSK (1600/3200 bps) and 4FSK (3200/6400 bps).

notpike, 2022-01-10:
> I've put a bounty on this protocol support. It's more common than POCSAG in the US as well, as far as I can see. What I'd like to see from the bounty is the abilty to receive and decode FLEX like [multimon-ng](https://github.com/EliasOenal/multimon-ng) ideally for both 2FSK (1600/3200 bps) and 4FSK (3200/6400 bps). I don't know of any available FLEX transmitting software so for testing one would need to be in an area with FLEX signals, but if not it might be possible to collect some sampled signals off the air for replay via sdr.

Hi!

I'm in possession of pager testing equipment that's capable of transmitting FLEX. I don't own any FLEX receivers however I'll be willing to help in the near future. :3

ckuethe, 2022-01-10:
https://github.com/unsynchronized/gr-mixalot can help you transmit.

epvuc, 2022-01-10:
On 2022-01-10 13:36, NotPike wrote:
> Hi!
>
> I'm in possession of pager testing equipment that's capable of
> transmitting FLEX. I don't own any FLEX receivers however I'll be
> willing to help in the near future. :3

Multimon-ng on linux will receive FLEX accurately, and I think there are 
a bunch of windows things, too, lus the gnuradio module.  Also I'm sure 
off the air samples could be made available.

Micha-123, 2022-01-11:
I can record an "stream" with the hackrf and send it to you, in NL is FLEX
used for the liveP2000.  (Look at live p2000)

ArjanOnwezen, 2022-01-21:
Moved from #474
Volleyoghurt commented 23 minutes ago • 
In the Netherlands we use Flex (P2000) for emergency services.
Is it possible to add this feature to the mayhem software.

The decoding of the software is already been done for Linux.
https://github.com/EliasOenal/multimon-ng

More information about the protocol.

http://jelmerbruijn.nl/motorola-flex-p2000-decoding/
Frequency is 169,650 MHz
I don't have the knowledge to convert it to the mayhem software.

The best possible solution is the have a RX and a TX option.

Thanks!

ArjanOnwezen, 2022-01-28:
As mentioned in #483. https://github.com/craigshelley/multimon-ng/blob/master/demod_flex.c might be a good start.

eried, 2023-12-05:
Hi! anyone who added money to this bounty, please try to open a dispute on paypal, check -> https://github.com/bountysource/core/issues/1586

closed 2023-12-05 by eried