portapack-mayhem/mayhem-firmware #319: Limit 2 MHz.

Merianna, 2021-03-06:
Hi guys, I have to solve a problem, I need to display some signals at 1Mhz and at 500Hz, but hackrf does not show below 2 MHz.
How can I do?

Thanks

laneends, 2021-03-06:
Use an upconverter, such as the noelec ham it up.

Merianna, 2021-03-06:
Thanks laneends, but I needed a solution integrated with the portpack.  Is it not possible to act via firmware?

laneends, 2021-03-06:
This is not possible, it's a hardware issue, hacked is 1mhz to 6 ghz

Merianna, 2021-03-06:
But why can't I get a band resolution of at least 1MHz?

laneends, 2021-03-06:
Ah, I misunderstood.

GullCode, 2021-03-06:
@laneends  no I think you understood correctly, but @Merianna  is asking one more question.

For frequencies below 1/2MHz you have not other choice but to ham it up like @laneends said.

For the bandwidth, as far as I know and if it is to look at a signal, then Looking Glass app can show it.

Merianna, 2021-03-06:
Thanks for your patience.
I'll explain what I would like to do but I still don't know how to proceed;

I would like to scan the frequencies in a range from 100 to 500 Mhz in the 1MHz BW to find all the 500 Hz signals that have more than 20 dbm of power.
I tried with the "SCANNER" app but this is impossible.

How can I do?

GullCode, 2021-03-06:
Range from 100 to 500 Mhz => ok possible. Manual Search or a range preset in a file. Actual scanner app would not load that much frequencies, but the Search app in test drive would.
500Hz => ou mean the step ? no AM or FM signal is that wide.
>20DB => should be possible with adjustments on  squelch value.

Merianna, 2021-03-06:
500mhz i mean the width of the signal channel
The preset search range in a file
The signal is CW.

Look in the image what I need to look for.

[image]

GullCode, 2021-03-06:
CW as AM - CW ? in that case there is a setting like this in both Search and Scanner.
All AM modulations are using these settings:
receiver_model.set_sampling_rate(2000000);
receiver_model.set_baseband_bandwidth(2000000); 
Then a tap filter is applied for each different modulation.
So I believe just using AM/CW and playing with squelch should be enough.
Using the Scanner you may be able to search for a maximum of a few hundreds frequencies. Using the Search app you can search for as much frequencies there are in range like in (end - start) / step = nb searched frequencies.
You may need to play with squelch and step to see what's fitting.

Merianna, 2021-03-07:
Thanks for your help.  
I tried to come you advised me but it does not seem possible to capture these signals that you see in the image.

It is important for me to find a solution and if the device was able to capture signals like these I would be interested, if you are also available by making a donation, develop an app that I would have created.

[image]

eried, 2021-03-08:
You can attach a money bounty directly to [this issue](https://www.bountysource.com/issues/96947763-limit-2-mhz?utm_campaign=plugin&utm_content=tracker%2F182166508&utm_medium=issues&utm_source=github) but I do not think @GullCode is hunting from bounties 🤣 (he himself donated a lot to other bounties already from his pocket)

Merianna, 2021-03-08:
Thanks for the ideas, could you recommend someone to contact to develop the app?

laneends, 2021-03-09:
It would appear that you just need to use a different software with a smaller bandwidth, perhaps SDR#. Alternatively, you can purchase a portapack and download the mayhem firmware. This has a CW setting.

Merianna, 2021-03-09:
I opened a request https://github.com/eried/portapack-mayhem/issues/322 but I figured out how to make a bounty :(
Thanks

eried, 2021-03-10:
I explained how to add the bounty on #322 ;)

closed 2021-03-10 by eried