portapack-mayhem/mayhem-firmware #320: Portapack H2 No. 1 VS No. 2

iz2fly, 2021-03-06:
I bought a new Portapack No.1 different from the previous No.2
In the latter model, that is the No.1, I highlighted with the use of the microphone an accentuated background noise that is not present with the No.2.
I have tried several microphone capsules and different software versions, but the problem remains. Without inserting the microphone there is no background noise. Probable hardware problem, I believe that without a datasheet it is difficult to find a  
[image]

GullCode, 2021-03-06:
Maybe it's badly grounded ? Have you checked the soldering ?

iz2fly, 2021-03-06:
Maybe some component is missing? I don't have another to compare.
Video at the bottom of my website page. 
http://iz2fly.altervista.org/hackrf_one/

[image]

eried, 2021-03-08:
I have the style 2, so I cannot compare. But this is quite nice info for the wiki about this "illegal" H2's

ArjanOnwezen, 2021-06-24:
This is really a hardware issue. For questions regarding hardware, please ask it on Discord.

closed 2021-06-24 by ArjanOnwezen