portapack-mayhem/mayhem-firmware #321: Sound through phone jack not working with 1.3.1 but with 1.3.0, why?

port513, 2021-03-08:
When the unit was delivered I had Mayhem 1.3.0 installed, thought it would be a nice idea to have the latest so I flashed it and after that I get no sound from phone jack at all. Why?

I have Portapack H2+ with battery installed but with no internal speaker connected.

And one other weird thing, unit refuse to startup as long as my SD-card is inserted. I have to boot it without SD-card and insert it after it have started up. Why does it do that? it can read the SD-card without problem but not boot while it's inserted at boot.

port513, 2021-03-08:
Solved this myself, I needed this version of Mayhem https://www.mmdvm.club/usr/uploads/2021/03/232153741.zip

closed 2021-03-08 by port513

eried, 2021-03-09:
Ohh this is crappy. Can you share the vendor who sold you this unit so other people can avoid it? I will add this issue to the warning:
[image]

port513, 2021-03-10:
It does work with 1.3.1 but the custom variant.

https://m.banggood.com/3_2-inch-Touch-PORTAPACK-H2-+-HACKRF-ONE-SDR-Radio-with-Firmware-+-0_5ppm-TCXO+Battery-+Case-C5-019-p-1771648.html?rmmds=search&act_poa=SKUG59756&cur_warehouse=CN