portapack-mayhem/mayhem-firmware #322: Search for signals

Merianna, 2021-03-09:
Hi, Can anyone help me?  

I need a Portpack Hackrf device app.
This APP must be the only one present in the device and will start when it is turned on.

The function of this APP is to scan a range of frequencies looking for CW signals that have a certain intensity, (for example -20 dBm) with a channel width that I will choose.

The settings; frequency range, signal strength and channel width will be configured via a file placed in the SD.

If during the scanning of the frequencies corresponding signals are found, these must be stored in some boxes placed in a grid on the screen and can be recalled by touching one of them.

When I check one of these boxes I will be able to see the + - signal strength through colored bars, which will increase as I get closer to the source.

For details I will explain later.

Thanks

This is the type of signal I would like to look for.

[image]

eried, 2021-03-10:
For adding a bounty you should just read the instructions on the main readme, but now you can even click under your post in "Post a bounty on it!" link that was added automatically. Remember this is just to try to get interested people to work on this, but it is not an assurance of a result.

Merianna, 2021-03-10:
I added a bounty of $50 via paypal but it doesn't appear on the site.

euquiq, 2021-03-10:
Just my two cents:

This is very similar to "looking glass" where you can set the lower / upper freq you want to scan, and visually filter the result for strong signals.

It should be "easy" to add a space, below the screen, where the "strong" signals are printed out (it wouldn't be 100% precise, tho). 

Having said that, here comes my "negative" comments:

This seems to be a "custom" work. Per your asking, it is not Mayhem related, since you need the device to boot up directly into your app. I know I wouldn't do it for 50 usd ... nor 5000 usd. 

Now back to a last comment (but positive one):

It may be fun to add into settings / interface the capability for starting one specific app upon booting. Or even being able to select which app to automatically start on booting, based on the depressed direction buttons at booting time. (but never leave off the "mayhem" ecosystem).

Merianna, 2021-03-10:
Perhaps there is a misunderstanding, it was my intention to find someone to help develop a custom app.

eried, 2021-03-11:
Oh in that case you should withdraw your bounty, because the idea of bounties is only to shape the functionality in a way that can be added for everyone's benefit

ArjanOnwezen, 2021-11-28:
This is a community project. For customized firmware, find (and pay) a good developer.

closed 2021-11-28 by ArjanOnwezen