portapack-mayhem/mayhem-firmware #324: Problem upgrading the firmware

TomToy621, 2021-03-09:
**What is happening?**
Describe here why you are unable to upgrade the firmware. Before describing your problems, **do the following**: 
* Read: https://github.com/eried/portapack-havoc/wiki/Update-firmware
* Watch carefully: https://www.youtube.com/watch?v=_zx4ZvurgOs
* (if you are not in Windows) also check: https://www.youtube.com/watch?v=kjFB58Y1TAo

**Were you able to update the firmware before?**
Things might be confusing the first time, please check the video available on the link above.

**Can you try the upgrade with a different PC/Portapack/HackRF?**
If is possible, swap hardware and try again. Also, try different USB cables, even if the one you are using works fine for other purposes.

**Additional**
Add photos and videos of your procedure.

TomToy621, 2021-03-09:
Hello,

I think I watched the video carefully for the firmware update v1.3.1.
I've got an issue when i try to start my Portapack H2 HackRF, sometime it doesn't start at all, sometimes after few tries it starts after a while and sometime it starts very quickly.It also doesn't start at all with the micro SD inside before starting, it works when i add the sd card after the start...
Features i've tried for the moment seem to work awhen i succeed in starting my device and adding the SD card 

Could anyone tell me from where it can come from?

Thanks a lot for your help, have a good one

Tom

https://www.youtube.com/watch?v=FDNUBM1ruWk

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

eried, 2021-03-10:
It sounds like a hardware problem, how old is the unit?

TomToy621, 2021-03-10:
Hey @eried, thx for your reply!!I've just received the unit few dayz ago...

TomToy621, 2021-03-10:
As I won't get any refund or return possibility from BG, do you think it's a Hack RF 1 or Portapack H2 issue?Is there any possibility to check with for example a multimeter or any other device where the problem comes from?
Thanks

Tom

euquiq, 2021-03-10:
You can disassemble it (with care) and use the hack rf  alone, through USB, with your computer.

Also remember to never turn it on and transmit without an antenna attached to it. It is a fast way to burn the TX amp if used.

TomToy621, 2021-03-10:
Hey @euquid, thx, i did the firmware update with the hack rf dissasembled from the portapack as preconised...Will use the hack rf alone and perhaps later try to understand what's the issue

TomToy621, 2021-03-12:
thx, would be pleased to try this solution, but the link you gave me seems not to be working anymore, can you tell me the name of this firmware and if I can find it somewhere else?

TomToy621, 2021-03-12:
can I also try older versions of the mayhem?

port513, 2021-03-12:
> can I also try older versions of the mayhem?

If you have H2+ it will not work with older. You need this file https://www.mmdvm.club/usr/uploads/2021/03/232153741.zip

Just tried the link and it's still working as it should

I downgraded my H2+ and got a few hundred new bugs ;) So don't do that if you want to be sane this weekend ;)

zhang00963, 2021-03-24:
This seems to be caused by the firmware's failure to correctly recognize the H2 + audio chip,
You can modify the test as follows.

For / firmware / application/ portapack.cpp The file is compiled after modification.
It has been tested on ak4951en of H1 and wm8731 / wm8731s of H2. It can automatically identify the type of audio chip and drive it,
https://user-images.githubusercontent.com/14372128/111869569-2524f200-89bb-11eb-85c3-7efdffd77528.JPG

eried, 2021-03-25:
Hey @zhang00963 have you tried in other portapacks? with the old chip? Would you mind making a PR in github so everyone could test it?

zhang00963, 2021-03-26:
Hi @eried  I have portapack H1, which uses the audio chip of ak4951en.and portapack H2 + which uses the audio chip of wm8731s, and I try to replace wm8731s with wm8731 audio chip. After testing, portapack can work normally.
I think the problem is that in the init function, the I2C / I2S is not ready when the [portapack_cpld_config()] function is executed, resulting in the audio chip detection function not working properly,
I'll do some tests,Submit a PR.

eried, 2021-04-07:
> I'll do some tests,Submit a PR.

How are your tests going @zhang00963 ? It would be nice to release a new version with these soon so people do not need to worry about different versions

ckuethe, 2021-04-18:
As reported in #301, `portapack_nightly_202104180016_next.bin` allows my portapack to boot with an sdcard inserted.

ckuethe, 2021-04-22:
@TomToy621 try it with Mayhem 1.4.0 - it's working on my portapack.

ArjanOnwezen, 2021-11-28:
Looks like a duplicate of #346

closed 2021-11-28 by ArjanOnwezen