portapack-mayhem/mayhem-firmware #327: More a PSA: the seller linked on this page is selling the wm8731 version

nymda, 2021-03-14:
hey guys, kinda annoyed by this. i brought a hackRF + portapack combo from the seller listed here
[image]

when i received it i flashed the newest version of mayhem onto it and found that the headphone jack didnt work and i couldnt boot with an SD card inserted. Did some searching and found the wm8731 custom firmware and that seems to work fine. i dont mind using the custom wm8731 firmware but its irritating that the device listed on this page doesnt work with the normal firmware. 

if anyone has brought one recently did you get the same type?

eried, 2021-03-14:
Oh, not good! I will replace the link if this is the case, few people ordered that one (including me) and everyone got the non wm8731. Can you contact the seller and complain? Ask for a refund is possible

nymda, 2021-03-15:
i sent a message on aliexpress and the seller told me that he will be using the wm8731 chips from now on, i assume due to the shortage of ak chips after the fire. Can we get support for wm8731 chips in the normal version of the firmware? we are probably going to be seeing more of the second type of h2s from now on.

eried, 2021-03-15:
yes I agree, the crappy part is that they never made a clean PR with fixes and leave the problem to the "community". As last resource, if automatic detection cannot be done, we will add an option

nymda, 2021-03-15:
i dont see why automatic detection wouldnt be possible, surely the portapack knows what its own hardware is. it shows some info about it in the debug menu already

eried, 2021-03-15:
Yeah it might be possible, my point is that is quite extra work and testing from the "community" while the vendor who actually makes and sells the device just puts null effort. Only new users have wm8731, so none of the ones which can start testing a fix have both units...

nymda, 2021-03-15:
ahh true, didnt think of that. im happy to use my unit to test new versions if you need it.

eried, 2021-03-15:
Yeah thanks, maybe we need a new flag in discord to mark up the wm8731 users, in case we need testers

jamesshao8, 2021-03-17:
I have abundant ak4951 chip. Sure it's more expensive than wm8731 especially in these months, but it has more funtionalities.

eried, 2021-03-17:
more functionalities that affect the portapack? which ones?

jamesshao8, 2021-03-18:
If you go to debug menu, you will see there are much more registers in ak4951 than wm8731. ak4951 supports digital mic and some internal filters, which are controlled by these regs. Original designer of portapack upgraded the sound chip from wm8731(2015) to ak4951(2017), and he must have a reason for that.

Dirk275, 2021-03-18:
... then there should be supporting functions for wm8731 in firmware versions of 2015  ...
That's good news ...

zhang00963, 2021-03-20:
可对/firmware/application/portapack.cpp文件进行修改后编译，已在 H1的 AK4951EN 与 H2的WM8731/WM8731S 者2种portapack上测试OK，可自动识别音频芯片类型并进行驱动，
第一处：
-		if( audio_codec_wm8731.detected() ) {
-			model = PortaPackModel::R1_20150901;
-		} else {
-			model = PortaPackModel::R2_20170522;
-		}
+		model = PortaPackModel::R2_20170522;

第二处：
-	return (portapack_model() == PortaPackModel::R2_20170522)
-		? static_cast<audio::Codec*>(&audio_codec_ak4951)
-		: static_cast<audio::Codec*>(&audio_codec_wm8731)
-		;
+	if( audio_codec_wm8731.detected() ) 
+		return static_cast<audio::Codec*>(&audio_codec_wm8731);
+	else
+		return static_cast<audio::Codec*>(&audio_codec_ak4951);

trigger7, 2021-03-27:
As a potential buyer for an H2, this annoys me too...

I had a look at their suggested changes above, as well as firmware/application/portapack.cpp, and it will probably break something up...
The existing code uses the presence of a wm8731 to detect the hardware revision of the portapack ; there are 2 apparently, with changes on the CPLD config and the backlight (as well as the audio codec, obviously).
They basically disabled that and forced the newer (2017) revision. To fix their audio codec mess, they probably broke the 2015 revision of the portapack.

A proper fix would involve creating a third hardware revision, but we would need a way to differenciate between them not just based on the audio codec. I don't know if it's possible.
Or have different builds of the firmware and let the user flash the correct one...

On a side note, the wm8731 implementation seems quite incomplete compared to the ak4951... Not sure if/how it impacts the usability. Right now, I'm really put off and not sure if I should buy one...

eried, 2021-03-28:
Yes, it is not ideal but there will be a solution soon.

johnelder, 2021-03-29:
Just got my h2. It has the 4951. Let me know if you want a link to the seller.

trigger7, 2021-03-30:
@johnelder I may be interested, please.

Although my guess is that all chinese portapacks will come with a wm8731 from now on, and ak4981 ones are just leftover stock.
So the issue will need fixing anyway.

johnelder, 2021-03-30:
This is where I got it. Ordered last week.
https://www.ebay.com/itm/USA-PORTAPACK-H2-For-HACKRF-ONE-SDR-0-5ppm-TCXO-2100mAh-Battery-3-2-Touch-LCD/284062137089

zhang00963, 2021-03-30:
Hi @trigger7, Thank you for your feedback. Without considering the difference of audio chips, portapack does have two hardware versions, that is, 2015 version and 2017 version. The LCD control between the two versions is different, and the 2017 version adds GPS module control, audio chip control and other signals.

It is true that forced update to 2017 version of CPLD firmware will lead to black or white screen failure of portapack based on 2015 version of hardware.
But in the original code, "audio_codec_wm8731.detected()" function in "portapack_model()" function can't return normally, because when calling "portapack_model()" function, I2c and the audio clock are not ready, which will also force the CPLD firmware update to version 2017.

I will continue to analyze and try to distinguish the two hardware versions (2015 and 2017) from the software level,
The automatic identification of audio chips (including ak4951en / wm8731 / wm8731s) based on 2017 version hardware (including H1 / H2 / H2 +) has been tested and available. Please refer to the modification method I provided above.
Based on the 2015 version of hardware, I currently know H1, and I'm not sure whether there is a H2 / H2 + Design Based on this version.

At present, I have H1 / H2 / H2 + based on 2017 version hardware, and my H2 + is compatible with ak4951en and wm8731, which allows me to choose one from the other for welding
[image]

For the code implementation of wm8731,I think it's about done,
Because wm8731 does not have many control registers, it can only control the simple function of volume. The register table is as follows,
[image]

ohoeltke, 2021-04-02:
got mine last week from HamRadioshop Store (link from the readme.md).
Also contained the wm8731 :(

Now i'm in search of the custom firmware after "upgrading" to 1.3.3.

Hypsilofodon, 2021-04-02:
1.3.3  ? I think latest version is 1.3.1, you can download custom firmware from this link https://www.mmdvm.club/index.php/h2.html

Hypsilofodon, 2021-04-02:
> got mine last week from HamRadioshop Store (link from the readme.md).
> Also contained the wm8731 :(
> 
> Now i'm in search of the custom firmware after "upgrading" to 1.3.3.

1.3.3 ? I think latest version is 1.3.1, you can download custom firmware from this link https://www.mmdvm.club/index.php/h2.html

ohoeltke, 2021-04-02:
Sorry, typo. I meant 1.3.1
the link didn't load but found my way through archive.org to http://www.taozhengbo.com/mayhem_v1.3.1_FIRMWARE_WM8731.zip

zhang00963, 2021-04-17:
The program of automatic judgment of hardware has been basically completed, including R1 / r2 hardware.

Does anyone have a portapack based on R1 hardware design to facilitate testing?
Test [URL](https://github.com/zhang00963/portapack-mayhem/tree/test)

Since it takes about 5S to reprogram CPLD, you need to wait for a while. If it doesn't display, please restart the device and try again. After the normal display, you need to save the hardware version information in the settings.

[image]

sck-nogas, 2021-04-24:
I pulled your code and ran it on my H1... Hardware version showed "R1_20150901"

Commented here...

https://github.com/eried/portapack-mayhem/issues/346#issuecomment-826162072

bl4ckic3, 2021-06-15:
Weird thing is that they now mention that in this repo we now support both chips: 
[image]

And this is the from printed with the PortaPack in the link which is in the official page.

eried, 2021-06-15:
It kinda does because https://github.com/eried/portapack-mayhem/issues/346#issuecomment-861020907 but it seems to be still a problem for some units

trigger7, 2021-06-23:
Hey, back after a setback -- ordered a portapack that got lost in the mail, had to wait to get a refund, decided to order one from the seller linked on the main page.
The unit I received is not an H2.
It's an H2+ (which despite the name is a more basic version), without the extra ICs advertised on the aliexpress page, while still being 15-20$ more expensive then other H2+ sellers...
And of course it has a WM8731 chip instead of the AK4951, but I expected that (despite, again, the seller still advertising the AK chip on his product page).

My advice : go for the cheapest one you can find, there is no point trying to get the "better" hardware as, in the end, you'll get a random version anyway...

teixeluis, 2021-06-23:
Is there any shortcoming in having the WM8731 instead of the AK4951? There are some improvements/fixes which are not in the current release, but in functional terms is there anything we miss out on, for having the first instead of the latter?

trigger7, 2021-06-25:
Overall, the AK4951 is a better chip, but it probably does not make any difference in this use case :)

I'm more bummed out by the fact the seller is sending a different portapack that the one he's advertising.
While charging 20$ more than other aliexpress shops.
Probably because he's cashing on the extra traffic this github page is generating for him.

ArjanOnwezen, 2021-11-28:
Looks like a duplicate of  #346

closed 2021-11-28 by ArjanOnwezen