portapack-mayhem/mayhem-firmware #328: About portapack's touch, there are some problems

zhang00963, 2021-03-20:
I tested the hardware of H1 / H2 and found that the lower left corner of the screen could not be touched
As shown in the picture
H1：
[image]
H2：
[image]
Does anyone know the reason?

Radiomix2000, 2021-03-20:
Provide more information. What release?

zhang00963, 2021-03-20:
I'm on portapack H1/H2/H2+ hardware.
The firmware tested in v1.3.0 / V1.3.1 has this problem.
@Radiomix2000

euquiq, 2021-03-20:
AFAIK this is due to bad quality hardware, on "clone" portapacks. Mine has that problem. I fixed it by adding a "sensitivity selector" to the code, inside the touch calibration app. If you are on Discord server chat, please contact me, and I can give you a test firmware for you to try and see if it solves your situation too.

GullCode, 2021-03-21:
@euquiq why aren't you providing it in the stable releases ?

euquiq, 2021-03-21:
AFAIK the current policy for Mayhem is not to include "fixes" for bad / faulty / closed design firmware (like this problems with the cloned chinese portapacks ... or the new portapacks based on the old - lame-ish audo chip)

GullCode, 2021-03-21:
I understand the statement, but I also think it's not very user friendly even if it is the manufacturer's fault for not providing PRs for new hardware / bad designed screens.

Radiomix2000, 2021-03-21:
> 
> 
> I'm on portapack H1/H2/H2+ hardware.
> The firmware tested in v1.3.0 / V1.3.1 has this problem.
> The area cannot be touched as shown in the picture.
> @Radiomix2000

My one works good without issues. Seems to be hardware issue

eried, 2021-03-22:
I am gonna close this issue for now, because it seems to be a faulty unit but

closed 2021-03-22 by eried

zhang00963, 2021-03-22:
After testing, it is found that this area needs a large touch area to recognize, and a small area of click can not be recognized normally. Testing the four signals of the touch screen, it is found that one of the signals seems to be abnormal. The touch function is realized by the cooperation of mcu-adc and 5m40ze64c5,  I am checking whether there is any abnormality

zhang00963, 2021-03-22:
According to the NXP manual, the accuracy of MCU ADC is 10 bit, and the touch screen power supply control is realized by CPLD. This design seems not ideal. When the touch screen boundary area or CPLD control is not synchronized, ADC may not collect weak voltage signal, resulting in screen failure
[image]
[image]

I checked the internal block diagram of some touch screen control IC. The accuracy of ADC is 12 bit, and the power supply control of ADC sampling is controlled by the IC.
[image]

ohsix, 2022-02-04:
@eried can you advise on where I need to look to test this hypothesis?

eried, 2022-02-04:
How common is this in the newer units? (I would add some debugging values to the screen to start testing, but my unit does not has this issue)

ohsix, 2022-02-04:
can't speak to that, but mine has the exact same problem, edges of the void are in the same place and everything.

couldn't make total sense of the schematics and how the cpld is working to attempt to fix it myself so i was looking for ways to prove that it was a timing issue

changing the gnd/ref order proves 2 things:
1. the affine transform is what makes the lcd seem to sort of work 
2. the square void is just a timing issue

the way the algorithm works for fixing up touch points is really clever, it doesn't just cause this void but it makes the sensitivity vary in different regions of the screen, making it really hard to use

if it was just a matter of the bottom corner not really working i'd have to admit it wasn't a big deal

ArjanOnwezen, 2022-02-04:
@ohsix  and others. Is it model H2+ and/or do you have some more info about the hardware (sxreen, etc.)?

ohsix, 2022-02-05:
front text says H2+ yea. back says v3.4b.

[image]
[image]
[image]
[image]

and the hackrf clone it came with

[image]

ArjanOnwezen, 2022-02-05:
@ohsix  thanks @here do other with similar issues have the same device?

nonobzh29, 2022-12-15:
I have the same issue with a H2+ PCB v3.6 on a genuine HackRF with firmware 1.5.4
The bottom left of the screen is not responsive

snoehmer, 2023-05-02:
Hi,
first of all, thanks for the great work you are doing!
Unfortunately I am also having the same issue on a H2+ plus HackRF One combo I bought on Aliexpress.
The H2+ PCB is also marked with v3.6, and the HackRF One is also a clone (with the silk screen on the PCB looking different than shown in the above pictures).
I am running firmware 1.6.0.
I also tried with a genuine HackRF One, but the behaviour is the same.

Is it planned to include the touchscreen sensitivity setting ([this pull request](https://github.com/eried/portapack-mayhem/pull/147)) into the master branch / FW release at some point?
I don't know how many people are affected, but it would be great if the threshold could be adjusted easily.

vmakeev, 2023-05-24:
I agree with the author of the previous message. I have a version of the device H2+R4, the same problems with the lower left corner of the touchscreen (reacts only to pressing with a finger, but not with a stylus)

zxkmm, 2023-05-26:
not hardware issue. fixed. test binary: https://discord.com/channels/719669764804444213/722101917135798312/1111495356023320648