portapack-mayhem/mayhem-firmware #330: Again, about internal microphone init

Radiomix2000, 2021-03-22:
Hello, guys,
some times ago James Shao wrote his code to init internal microphone. He had his own fork which was not attached to any repo. So it was impossible to make pull request and add this code to main Eried/portapack repo.
I made fork of main repo on 29.09.2020, and copied James' code to my fork, which is attached to main Eried repo.
   The changes in code can be seen here:
https://github.com/Radiomix2000/portapack-mayhem/commit/9fb7678932d689454df4f620df17a59c8c9bdc8a
There is also pdf instructions, how to connect internal mike to pcb on portapack H1. As I know, my colleagues were able to make the same mod on H2 (suppose there are testpoints 4, 5, 6 on H2 pcb too). So I think this mod should be useful.
   Any chances to see this changes in main Erwin's repo?
I also need help to understand one bug in mic init code: if to enable RECEIVE in microphone app, the internal mic stops to work and portapack switches back to external 3.5 mm plug. When you disable RECEIVE, internal mic works again.
   Any comments are welcome.

eried, 2021-03-22:
Yes, I have integrated other functionality in that way in the past, but the problem in this case is that behaviour is only useful in H1, on H2 there is automatic switching, so what do you think?

Radiomix2000, 2021-03-22:
hmm, if behavior is different for H1 and H2, maybe it's good idea to make some setting in Settings menu, where user chooses, what version he has, H1 or H2, and then to handle mic initialization, according to chosen version?
..stop!, what do you mean under _automatic switching on H2_? Do I understand right, that H2 already has internal mic? And when you connect accessory, the internal mic is disconnected?

eried, 2021-03-22:
This is only for mic? H2 does not comes with internal mic, I thought it was the audio in general so maybe it makes sense to integrate

Radiomix2000, 2021-03-22:
Ok, so H1 and H2 are without internal microphone.
Yes, this code is only for mic. Audio IC, installed in both H1 and H2 is AK4951. This IC has two inputs. One is connected to 3.5 mm plug, and the second input is not used. So, we connect the internal microphone to this input. Switching is implemented in code, by setting corresponding registers of AK4951.

I just asked my colleague who has H2 about modification in H2. He will send me pictures of PCB with connections. So I will make instructions how to connect internal mike for H2 too. If necessary, H1 modification instruction pdf is in my repo.

Radiomix2000, 2021-03-22:
This is instruction, how to connect internal microphone on Portapack H2
[Internal microphone modification for HackRF Portapack H2.pdf](https://github.com/eried/portapack-mayhem/files/6185576/Internal.microphone.modification.for.HackRF.Portapack.H2.pdf)

eried, 2021-03-23:
Very good, maybe then you can do a PR with the modifications and we need to write a wiki page with this

gullradriel, 2023-05-10:
Closed due to inactivity

closed 2023-05-10 by gullradriel