portapack-mayhem/mayhem-firmware #335: How do I add a freq. snap to the Calls app?

fossum, 2021-04-12:
Forgive me if this isn't the right venue for this question, but I'm not sure where else to go.

When using the Calls app, I'd like to snap to 2.5KHz. This doesn't seem possible in the UI or the SD files. I'm wondering if I need to make an improvement to the app.

Does anyone know if there's something short of a feature enhancement that can solve this?

jwetzell, 2021-04-12:
those option are in the [header file](https://github.com/eried/portapack-mayhem/blob/4c50401df24924d63dbeeacfe58640071794f69a/firmware/application/apps/ui_search.hpp#L214) should just be a matter of adding 2.5Khz as an option to that OptionsField

fossum, 2021-04-13:
That worked wonderfully. Is there some design guide for stuff I'm adding? Or should I just submit a PR and see what happens?

jwetzell, 2021-04-13:
There is a discord for developments (amongst a lot of other conversations) but I would say for a simple probably just one line change no reason to send out the firmware and have everyone test a simple PR will do and I don't see why it wouldn't be easily mergeable

fossum, 2021-04-17:
Created #341

fossum, 2021-04-17:
> There is a discord for developments (amongst a lot of other conversations)

Got a link to that? I can't seem to find it through a search.

eried, 2021-04-19:
> Got a link to that?

It is on the main page, in the badges as "chat" or in the text below -> https://github.com/eried/portapack-mayhem#how-can-i-collaborate

closed 2021-04-21 by fossum