portapack-mayhem/mayhem-firmware #340: Portapack stuck in HackRF mode

EwartM, 2021-04-15:
My portapack is stuck in HackRF mode. See [this video](https://youtu.be/UOGUOvfgnso)
Mayhem has booted intermittently but I can't get it to show up now. 

Have tried:
flashing portapack-h1-fimware, portapack-h1_h2-mayhem, hackrf_one_bin, hackrf_one_dfu
disconnecting external and internal batteries
cleaning header contacts
resetting and all buttons in combination
re-seating the portapack on the HackRF

It's an original HackRF One with a clone PortaPack

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f61
Board ID Number: 2 (HackRF One)
Firmware Version: git-ef5b907 (API:1.03)
Part ID Number: 0xa000cb3c 0x00574752

Anything else I can try?

eried, 2021-04-16:
Can you flash portapack-h1_h2-mayhem.bin and share the output of the flashing process here? I am guessing you are not flashing anything because another error, so you are just keeping that hackrf only firmware

EwartM, 2021-04-16:
─$ hackrf_spiflash -w portapack-h1_h2-mayhem.bin
File size 873764 bytes.
Erasing SPI flash.
Writing 873764 bytes at 0x000000.
                                                                             
─$

EwartM, 2021-04-16:
I ran spiflash with verbose output and it looks like it completes without errors. 

...
Writing 256 bytes at 0x0d5400.
Writing 36 bytes at 0x0d5500.
                                                                             
─$

eried, 2021-04-16:
mmm I am not sure, what I would do is to try with the windows tools that go next to the firmware file from the release zip. Maybe your hackrf tools are old, or different?

EwartM, 2021-04-17:
Thanks for the help but I can't seem to get the Windows tools working. I've also just noticed that I dropped a blob of solder on the screen connector somewhere along the line. Tried removing it with hot air and solder wick but no dice - some pins are still connected together. I'll have to give up for now until I have more time to tinker.

closed 2021-04-17 by EwartM

eried, 2021-04-19:
What is wrong with the Windows tools? which kind of message do you get? From the zip file use the built-in ones. Those are proven to work for everyone.