portapack-mayhem/mayhem-firmware #342: Adding squelch in audio receiver

dragonis41, 2021-04-19:
**Description of the functionality**
Adding squelch in audio receiver for cutting the audio when no signal is received or to activate the audio only when a certain dB is reached.

**Additional**
This functionality exist in the scanner app for finding a signal.

KB6mcc, 2021-04-20:
Hi, Audio receiver does have the SQ function?  You will see it when just moved the cursor to the demodulation selection.

dragonis41, 2021-04-20:
Hi, tell me if I'm wrong but the NFM mode have a SQ function but not AM nor WFM

KB6mcc, 2021-04-20:
Oh....not for AM or WFM for sure..

teixeluis, 2021-05-31:
If it's an rssi based squelch, it shouldn't be too challenging to "port it" to AM/SSB as well (albeit for SSB such a squelch is only so much useful..).

IW3IPS, 2021-12-06:
AM squelch is useful for Air Band or CB reception.

Laidukas, 2024-09-25:
I would really appreciate squelch in AM demodulation as it is main modulation type in Air band and CB radios. I bought mine SDR hardware just for those bands.

NexGen-3D-Printing, 2025-01-09:
This would be a nice feature, and more arcuate filtering for AM as well.

zapfs, 2025-06-21:
It would be great if someone could add this feature. I‘m also listening to airband. T