portapack-mayhem/mayhem-firmware #343: Receiving failure.

Downisup410, 2021-04-20:
Anytime I try to use the receiver feature and attempt to record, the screen freezes and the only way to get it working again is to turn it off and back on. It doesn't gather any data either.

eried, 2021-04-20:
Has that ever worked for you? is this after a firmware upgrade? does it work in hackrf mode receiving with the computer?

gerryk, 2021-04-28:
Are other features that use the SD card working? Can you view the SD card using FileManager?

Downisup410, 2021-04-29:
Yes there are other features that work with it. I can capture and replay,
also I can view the files that I put on it that the firmware called for.
But for some reason it freezing when I try to record in receive mode

Downisup410, 2021-04-29:
No it hasnt. Everything else seems fine. Hackrf mode works fine. Everything
else works fine.

eried, 2021-04-29:
It can be different things, can you try other sd cards, while connected to usb with different usb cables and record a video on how it freezes?

Downisup410, 2021-04-29:
I will try the different sd cards here in a few and I will let you know.
Heres a short video of what happens

https://streamable.com/zyiff8

eried, 2021-04-30:
Can you record in the Capture app?

Downisup410, 2021-05-01:
Yes i can record in the capture feature.

eried, 2021-05-02:
Very strange then... But do you get new C16 files when recording? I would guess the problem is with the folder structure in the SD card then

ArjanOnwezen, 2021-05-28:
@Downisup410 Can you try it with the latest version (1.4.0) of Mayhem?

Downisup410, 2021-06-04:
I am pretty sure that is what I have. When was it released? and do any of you all know anything about a yardstick one?

ArjanOnwezen, 2021-06-04:
@Downisup410  just check version number bottom left when you switch on your PortaPack. That should give you the final verdict..

ArjanOnwezen, 2021-11-09:
Hi @Downisup410  without further info, it's impossible to diagnose, let alone solve the issue.
Closing issue. If you have more detailed info, please open another issue

closed 2021-11-09 by ArjanOnwezen