portapack-mayhem/mayhem-firmware #344: Receivers -->ERT Meter

KB6mcc, 2021-04-20:
New to this hobby and not sure it is a bugs or not.

ERT Meter able to captured the devices near by. However it does not show the captured devices information on the display but on to the log file "ert.txt" ???

ArjanOnwezen, 2021-07-27:
@KB6mcc Is it possible logfile contains all raw data and screen only shows valid packets?

KB6mcc, 2021-08-08:
Hi, Attached is the screenshot of the display and log file when the portpack using ERT
[image]
[image]

KB6mcc, 2021-08-08:
I will get the other info when I able to received a good working Meyers..

KB6mcc, 2021-08-08:
This attach for TPMS.  the screen not showing anything but the log file capture data...
[image]
[image]

KB6mcc, 2021-08-08:
More ERT data and nothing showed on display.
[image]

KimIV, 2021-11-02:
На экране показываются только текущие данные, а в журнале можно увидеть прошлые. Например, по TPMS Ваше фото с пустым экраном датируется 08.08.2021, а в журнале записи за предыдущий день, то есть 07.08.2021. У меня в этом плане всё ок. Прошивка 1.2.1.

[image]

ArjanOnwezen, 2021-11-02:
> На экране показываются только текущие данные, а в журнале можно увидеть прошлые. Например, по TPMS Ваше фото с пустым экраном датируется 08.08.2021, а в журнале записи за предыдущий день, то есть 07.08.2021. У меня в этом плане всё ок. Прошивка 1.2.1.
> 
> [image]

Could you please write in English? So all of us could understand and reply. Thank you.

KimIV, 2021-11-02:
https://translate.google.com/?sl=auto&tl=en&text=%D0%9D%D0%B0%20%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%B5%20%D0%BF%D0%BE%D0%BA%D0%B0%D0%B7%D1%8B%D0%B2%D0%B0%D1%8E%D1%82%D1%81%D1%8F%20%D1%82%D0%BE%D0%BB%D1%8C%D0%BA%D0%BE%20%D1%82%D0%B5%D0%BA%D1%83%D1%89%D0%B8%D0%B5%20%D0%B4%D0%B0%D0%BD%D0%BD%D1%8B%D0%B5%2C%20%D0%B0%20%D0%B2%20%D0%B6%D1%83%D1%80%D0%BD%D0%B0%D0%BB%D0%B5%20%D0%BC%D0%BE%D0%B6%D0%BD%D0%BE%20%D1%83%D0%B2%D0%B8%D0%B4%D0%B5%D1%82%D1%8C%20%D0%BF%D1%80%D0%BE%D1%88%D0%BB%D1%8B%D0%B5.%20%D0%9D%D0%B0%D0%BF%D1%80%D0%B8%D0%BC%D0%B5%D1%80%2C%20%D0%BF%D0%BE%20TPMS%20%D0%92%D0%B0%D1%88%D0%B5%20%D1%84%D0%BE%D1%82%D0%BE%20%D1%81%20%D0%BF%D1%83%D1%81%D1%82%D1%8B%D0%BC%20%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%BE%D0%BC%20%D0%B4%D0%B0%D1%82%D0%B8%D1%80%D1%83%D0%B5%D1%82%D1%81%D1%8F%2008.08.2021%2C%20%D0%B0%20%D0%B2%20%D0%B6%D1%83%D1%80%D0%BD%D0%B0%D0%BB%D0%B5%20%D0%B7%D0%B0%D0%BF%D0%B8%D1%81%D0%B8%20%D0%B7%D0%B0%20%D0%BF%D1%80%D0%B5%D0%B4%D1%8B%D0%B4%D1%83%D1%89%D0%B8%D0%B9%20%D0%B4%D0%B5%D0%BD%D1%8C%2C%20%D1%82%D0%BE%20%D0%B5%D1%81%D1%82%D1%8C%2007.08.2021.%20%D0%A3%20%D0%BC%D0%B5%D0%BD%D1%8F%20%D0%B2%20%D1%8D%D1%82%D0%BE%D0%BC%20%D0%BF%D0%BB%D0%B0%D0%BD%D0%B5%20%D0%B2%D1%81%D1%91%20%D0%BE%D0%BA.%20%D0%9F%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0%201.2.1.&op=translate

KimIV, 2021-11-05:
Here, I checked it on version 1.4.0. Four sensors from a Ford Kuga car. The last two lines are, apparently, just rubbish.

[image]

ArjanOnwezen, 2021-11-05:
> Here, I checked it on version 1.4.0. Four sensors from a Ford Kuga car. The last two lines are, apparently, just rubbish.
> 
> [image]

Thanks for checking KimIV. The rubbish could be eliminated by using  checksum or so, if supported by protocol

euquiq, 2021-11-05:
In short, then, this issue can be closed

closed 2021-11-08 by eried