portapack-mayhem/mayhem-firmware #346: Problem upgrading firmware to 1.4.0 release

sck-nogas, 2021-04-23:
**What is happening?**
On my HackRF with PortaPack H1 (Green 20140826), I've been compiling/running 1.3.1 on it wonderfully. When I saw the announcement for 1.4.0 I decided to git update/compile/update to it. So, I did the usual compile steps on my Mac...

`cd ~/src/portapack-mayhem/ && git pull && rm -rf build && git submodule update --init --recursive && export PATH=~/bin/gcc-arm-none-eabi-9-2020-q2-update/bin:${PATH} && mkdir build && cd build && cmake .. && make firmware && hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin ; fi`

And it booted to a black screen. :( 

Hmm, maybe it's my compile, let's pull and use the released firmware first...

1. Put my Hackrf/PortaPak H1 in DFU mode, and can see it in lsusb/ioreg and dfu-util

```
iBiko:mayhem_v1.4.0_FIRMWARE sck$ ioreg -p IOUSB  -w 0 -r -n LPC@00320000
+-o LPC@00320000  <class AppleUSBDevice, id 0x1002de926, registered, matched, active, busy 0 (1 ms), retain 12>
{
  "sessionID" = 98877689078991
  "idProduct" = 12
  "iManufacturer" = 1
  "bDeviceClass" = 0
  "bMaxPacketSize0" = 64
  "bcdDevice" = 256
  "iProduct" = 2
  "iSerialNumber" = 3
  "bNumConfigurations" = 1
  "Bus Power Available" = 250
  "USB Address" = 17
  "Built-In" = No
  "locationID" = 3276800
  "bDeviceSubClass" = 0
  "bcdUSB" = 512
  "USB Product Name" = "LPC"
  "PortNum" = 2
  "non-removable" = "no"
  "kUSBSerialNumberString" = "ABCD"
  "bDeviceProtocol" = 0
  "AppleUSBAlternateServiceRegistryID" = 4297976100
  "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
  "Device Speed" = 2
  "USB Vendor Name" = "NXP"
  "idVendor" = 8137
  "kUSBCurrentConfiguration" = 1
  "IOGeneralInterest" = "IOCommand is not serializable"
  "kUSBProductString" = "LPC"
  "USB Serial Number" = "ABCD"
  "kUSBVendorString" = "NXP"
  "IOClassNameOverride" = "IOUSBDevice"
}
iBiko:mayhem_v1.4.0_FIRMWARE sck$ dfu-util -lv
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1fc9:000c] ver=0100, devnum=31, cfg=1, intf=0, path="0-3.2", alt=0, name="DFU", serial="ABCD"

iBiko:mayhem_v1.4.0_FIRMWARE sck$
```

2. Now, update the DFU with the HackRF DFU version from the latest release of Mayhem

```
iBiko:mayhem_v1.4.0_FIRMWARE sck$ dfu-util --device 1fc9:000c --download ./hackrf_one_usb.dfu --reset
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download[=========================] 100%35508 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
iBiko:mayhem_v1.4.0_FIRMWARE sck$
```

3. Now we can see the vanilla hackrf firmware in lsusb/ioreg

```
iBiko:mayhem_v1.4.0_FIRMWARE sck$ ioreg -p IOUSB  -w 0 -r -n "HackRF One@00320000"
+-o HackRF One@00320000  <class AppleUSBDevice, id 0x1002deb01, registered, matched, active, busy 0 (1 ms), retain 12>
{
  "sessionID" = 99725051152466
  "idProduct" = 24713
  "iManufacturer" = 1
  "bDeviceClass" = 0
  "bMaxPacketSize0" = 64
  "bcdDevice" = 260
  "iProduct" = 2
  "iSerialNumber" = 4
  "bNumConfigurations" = 1
  "Bus Power Available" = 250
  "USB Address" = 22
  "Built-In" = No
  "locationID" = 3276800
  "bDeviceSubClass" = 0
  "bcdUSB" = 512
  "USB Product Name" = "HackRF One"
  "PortNum" = 2
  "non-removable" = "no"
  "kUSBSerialNumberString" = "RunningFromRAM"
  "bDeviceProtocol" = 0
  "AppleUSBAlternateServiceRegistryID" = 4297976575
  "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
  "Device Speed" = 2
  "USB Vendor Name" = "Great Scott Gadgets"
  "idVendor" = 7504
  "kUSBCurrentConfiguration" = 1
  "IOGeneralInterest" = "IOCommand is not serializable"
  "kUSBProductString" = "HackRF One"
  "USB Serial Number" = "RunningFromRAM"
  "kUSBVendorString" = "Great Scott Gadgets"
  "IOClassNameOverride" = "IOUSBDevice"
}
iBiko:mayhem_v1.4.0_FIRMWARE sck$
```

4. Now we can push the old portapack firmware from the latest release of Mayhem

```
iBiko:mayhem_v1.4.0_FIRMWARE sck$ hackrf_spiflash -w portapack-h1-firmware.bin
File size 325092 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 325092 bytes at 0x000000.
iBiko:mayhem_v1.4.0_FIRMWARE sck$
```

5. This works, (after hitting the reset button), can boot Portapak and can see that I'm running nightly-1fe2d7b So, next, enter HackRF mode and update to 1.3.1 release of Mayhem.

```
iBiko:mayhem_v1.4.0_FIRMWARE sck$ hackrf_spiflash -w ../mayhem_v1.3.1_FIRMWARE/portapack-h1_h2-mayhem.bin
File size 873764 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 873764 bytes at 0x000000.
iBiko:mayhem_v1.4.0_FIRMWARE sck$
```

6. This works, (after hitting the reset button), can boot mayhem 1.3.1 and can see that I'm running v1.3.1. So, let's enter HackRF mode and update to 1.4.0 release of Mayhem.

```
iBiko:mayhem_v1.4.0_FIRMWARE sck$ hackrf_spiflash -w portapack-h1_h2-mayhem.bin
File size 903804 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 903804 bytes at 0x000000.
iBiko:mayhem_v1.4.0_FIRMWARE sck$
```

7. When I hit the Reset button, the screen shows white for a second and then a black screen...

**Were you able to update the firmware before?**
Yes, many times. 

**Can you try the upgrade with a different PC/Portapack/HackRF?**
Have tried 3 different cables and ports on my Mac

What do you want me to try next?
**Additional Details**
The cast of characters...
`iBiko:mayhem_v1.4.0_FIRMWARE sck$ shasum ./hackrf_one_usb.dfu ./portapack-h1-firmware.bin ../mayhem_v1.3.1_FIRMWARE/portapack-h1_h2-mayhem.bin ./portapack-h1_h2-mayhem.bin 
5e3cab8fea1f9e05a5b0dd790ef5e9a88b5ed65d  ./hackrf_one_usb.dfu
ef242587282402465232bd0cd8843dc2574c9c5d  ./portapack-h1-firmware.bin
b3c9863673e0fdf5588d2a77d893d36c19835be9  ../mayhem_v1.3.1_FIRMWARE/portapack-h1_h2-mayhem.bin
47202d2f782e95a5e3c30f0fad731efd1260c3a3  ./portapack-h1_h2-mayhem.bin
iBiko:mayhem_v1.4.0_FIRMWARE sck$ `

Misfitsman805, 2021-04-24:
I have this same issue using the pre-compiled 1.4.0 firmware release. So there is a problem somewhere in the firmware when using the H1 Portapack. Briefly shows a white screen and then goes to a blank black screen. I can still use the controls to put it back into hackrf mode though so it isn't locked up or anything like that. I'm running an Original HackRF board and an Original ShareBrained Technology PortaPack H1 20150901 Black Board.

rogueops, 2021-04-24:
Same here,I had to revert back to 1.1.1

gerryk, 2021-04-24:
I have a pretty early Chinese H1. Works fine with 1.4.0

rogueops, 2021-04-24:
What steps did you do? I have a portapack H1 and tried windows and linux
and both gave the same result and I had to restore and when i tried 1.1.1
it loaded with no issues. 1.4, just a black screen that flashes white once
when reset or when unplugged and plugged back in, on linux i did
hackrf_spiflash portapack-h1_h2-mayhem.bin, my version on hand was 1.1.1
which is why i went back to it, but I did see before 1.4 version 1.3.1 so I
downloaded that version and will try that version and report back, also
just redownloaded 1.4 in case it was a bad download

gerryk, 2021-04-24:
I just downloaded the release binary and SD card archive from github. I put the contents of the SD card archive on the SD card and used hackrf_spiflash to flash the firmware binary.

`hackrf_spiflash -w <firmware>`

rogueops, 2021-04-24:
just tested 1.3.1 and no issues, will test 1.4 tomorrow or should I say
today, its 5AM and I have yet to go to sleep yet

rogueops, 2021-04-24:
i tried 1.4 coming from 1.3.1 and no go, just a black screen, i am using
the pre-compiled releases. I update the sd card first, then the hackrf and
all i get is a black screen with 1.4

sck-nogas, 2021-04-24:
> I have this same issue using the pre-compiled 1.4.0 firmware release. So there is a problem somewhere in the firmware when using the H1 Portapack. Briefly shows a white screen and then goes to a [...]

How do you use the controls to put it into HackRF mode, if you can't see the screen? 

Just blindly, click "center" (to clear the spash screen), then click "down" five times, then click "right" once (to get to HackRf), and then click "center" twice (to select and then say "yes")? 

That was the steps on 1.3.1 to get to hackrf mode, doesn't work on 1.4.0 for me...

GullCode, 2021-04-24:
I think all the people here having a problem with that version may report the audio chipset they are using. Last changes may have break something on some devices.

sck-nogas, 2021-04-24:
Well, I have the official Portapack H1 Green board ([20140826](https://github.com/sharebrained/portapack-hackrf/blob/604389f8cd9118af489427ef7b6ea2bba3ca7947/hardware/portapack_h1/audio.sch#L354))with the WM8731 CODEC. 

If you have an official Portapack H1 Blue board ([20181029](https://github.com/sharebrained/portapack-hackrf/blob/1eeac6ae5d1a889c7932e021b20cfd5b030bb1c3/hardware/portapack_h1/audio.sch#L356)), you have the AK4951 CODEC

Not sure about the clones...

GullCode, 2021-04-24:
Maybe  a consequence of https://github.com/eried/portapack-mayhem/commit/d57edd5e25bed15278f933f927755ad7047693d7

sck-nogas, 2021-04-24:
> Maybe a consequence of commit [838451f](https://github.com/eried/portapack-mayhem/commit/838451f491aff1823d87d914b473b2112696e01c)

Well, I reset to 1.3.1, then did a quick git roll-back and rebuild from the commit just before that one to test this with the following commands...
`cd ~/src/portapack-mayhem/ && git reset --hard 2da482f393d56877d00bce293eb0bcd0f486bbcd && rm -rf build && git submodule update --init --recursive && export PATH=~/bin/gcc-arm-none-eabi-9-2020-q2-update/bin:${PATH} && mkdir build && cd build && cmake .. && make firmware && hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin`

And ....

The build finishes, creates the firmware, and installs? Can't really tell if it actually installed or not, since the version listed still says 1.3.1. But will try the next commit to see...

sck-nogas, 2021-04-24:
> Well, I reset to 1.3.1, then did a quick git roll-back and rebuild from the commit just before that one to test this with the following commands...
> `cd ~/src/portapack-mayhem/ && git reset --hard [...]

Tried again with...
`cd ~/src/portapack-mayhem/ && git reset --hard 838451f491aff1823d87d914b473b2112696e01c  && rm -rf build && git submodule update --init --recursive && export PATH=~/bin/gcc-arm-none-eabi-9-2020-q2-update/bin:${PATH} && mkdir build && cd build && cmake .. && make firmware && hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin`

And...

The build finishes, creates the firmware, and then after installing and reset... Boots to a black screen...

You're RIGHT!!! It is a consequence of commit [838451f](https://github.com/eried/portapack-mayhem/commit/838451f491aff1823d87d914b473b2112696e01c)

sck-nogas, 2021-04-24:
As a quick test, I did a git pull to the most recent version, and un-commented the lines 184, 185, 186, & 188 in [firmware/application/portapack.cpp](https://github.com/eried/portapack-mayhem/commit/838451f491aff1823d87d914b473b2112696e01c#diff-0bc451f7c724e9ede7c85511b1df169accbc10d07ce30781f21896568d9ac820) to see if just re-adding the check for the codec would work for me and...

I am now running a mostly 1.4.0 version... 👍

eried, 2021-04-24:
Oh, what if you leave the black screen for half a minute? it does never boot?

The author of that detection made comments about this:
https://github.com/eried/portapack-mayhem/issues/327#issuecomment-821854558

I do not have different versions to test :/

sck-nogas, 2021-04-24:
> Oh, what if you leave the black screen for half a minute? it does never boot?
> 
> The author of that detection made comments about this:
> [#327 (comment)](https://github.com/eried/portapack-mayhem/issues/327#issuecomment-821854558)
> 
> I do not have different versions to test :/

Well, I loaded the 1.4.0 release firmware and let it sit with a black screen for 2 minutes, then clicked "reset" and let it sit for another 2 minutes with a black screen. No change... :(

I then loaded my "tweaked" 1.4.0 where I un-commented lines 184, 185, 186, & 188 in firmware/application/portapack.cpp and with 4-5 seconds of power on, I could see the splash screen. But no information in the UI page to answer his question.

Next, I cloned zhang00963's fork from https://github.com/zhang00963/portapack-mayhem, switched to his 'test' branch and compiled a version of that, since the version of his code you referred to is different than the release version. Powered it on and saw a white screen for 5-8 seconds, then the splash screen! When to "Options"->"Interface" and the Hardware version showed "R1_20150901"

Misfitsman805, 2021-04-24:
> 
> 
> > I have this same issue using the pre-compiled 1.4.0 firmware release. So there is a problem somewhere in the firmware when using the H1 Portapack. Briefly shows a white screen and then goes to a blank black screen. I can still use the controls to put it back into hackrf mode though so it isn't locked up or anything like that. I'm running an Original HackRF board and an Original ShareBrained Technology PortaPack H1 20150901 Black Board.
> 
> How do you use the controls to put it into HackRF mode, if you can't see the screen?
> 
> Just blindly, click "center" (to clear the spash screen), then click "down" five times, then click "right" once (to get to HackRf), and then click "center" twice (to select and then say "yes")?
> 
> That was the steps on 1.3.1 to get to hackrf mode, doesn't work on 1.4.0 for me...
> 
> Update: When I do the blind click pattern listed above in 1.3.1 (the LEDs that light are...3v3 (green), 1v8 (amber),RF (red), USB (green), RX(off), Tx (off)
> 
> When I do the blind click pattern listed above in 1.4.0 (the LEDs that light are...3v3 (green), 1v8 (amber), RF (red), USB (off), RX(amber), Tx (off)

I have the splash screen disabled from when I had 1.3.1 So all I have to do is rotate the dial all the way to the right to scroll down to the end and then press the center button twice to get back to the hackrf mode.

zhang00963, 2021-04-25:
@sck-nogas Hello, thank you for trying，
Yes, in the previous code, the method of using audio chip to distinguish hardware version is not reliable, which will cause some portapack black screen or white screen,
I added the detection of the hardware version, but due to the test, it was not submitted synchronously with the detection function of the audio chip, which would cause the R1 hardware to have a white or black screen,
@sck-nogas @GullCode @rogueops @gerryk @eried 
Here to use R1 hardware friends to say sorry.

According to your test, this method of judging the hardware version should be effective. In addition, I don't particularly understand whether "ui_config" in "backup_ram" has default parameters. Does anyone know this?

I will submit PR later

GullCode, 2021-04-25:
@zhang00963 '"ui_config" in "backup_ram" has default parameters.' if it's coming from persistent memory there is no way to know what state a bit will have in persistent memory at first read.
Tests I've made showed that it's not a zeroed area, which means new values sometimes have something else than zero as a default.
The only actual way I know to do that is a to use a file on the sdcard: if not present, create it with default settings, else load settings from it.

zhang00963, 2021-04-25:
@GullCode Hello! 
 Yes, this parameter is from long-term memory. Do you know where this default parameter is in the source code?
 I need to consider that there is no memory card. According to my test, the value is not zero when the first power is on. Most of the bits are one. The default parameters should be written in the program, because some settings of the UI are not affected.

GullCode, 2021-04-25:
It's lying inside portapack_persistent_memory.cpp hpp.

But as I stated default value will differ from devices to devices.

The only actual way it to use a file on the sd card: if not present then default settings, save, and do the one time init proc.
Else load setting from file.

GullCode, 2021-04-25:
Can't we stick to models ? Aren't they consistent ?

zhang00963, 2021-04-25:
@GullCode ,  Hi!
I didn't see the default value of "ui_config" in "portapack_persistent_memory.cpp/hpp", but the way to get and set the  "ui_config" value in "portapack_persistent_memory.cpp/hpp".

Yes, R1 and R2 have different hardware. They use different CPLD programs

GullCode, 2021-04-25:
Because it does not have any default.

zhang00963, 2021-04-25:
@GullCode 
All right.., The existing judgment method should also be reliable. I'm going to analyze the reliability of configuring hardware version through SD card。

GullCode, 2021-04-25:
If it works it will be at least reliable :-)

eried, 2021-04-26:
> @GullCode
> All right.., The existing judgment method should also be reliable. I'm going to analyze the reliability of configuring hardware version through SD card。

The only problem with this solution would be that the new units usually ship without an SD card :/

I would prefer another solution (i.e. if UP and LEFT is held on boot, change the persistent variable to the alternative mode, something like that) but we should discuss this further

gerryk, 2021-04-26:
> UP and LEFT is held on boot

So before anyone takes this as an actual requirement, rather than just an example, this is hard to do on a H1.

zhang00963, 2021-04-26:
@eried  
Yes, I have considered this problem, and I find that SD card can't be recognized normally until the device is started for a while. I prefer to press OK key for a long time when the device is started to enter hardware detection mode.

zhang00963, 2021-04-26:
@eried 
According to the analysis, it is also unreliable to press the key, because the function of the key is supported by the CPLD program. It should be considered that the CPLD has no programming and the CPLD program is wrong.
[image]

eried, 2021-04-26:
OK, so maybe just have a new variable that counts failed boots? i.e. first increase a variable `boot_fail_count` and the moment the first key is pressed, subtract it from the same variable (so we know CPLD program was ok). If `boot_fail_count` > N switch to the alternative mode?

zhang00963, 2021-04-26:
@eried 
My current practice is to define 3bit in persistent memory, which is used to save hardware version, status flag, etc.
If version selection is enabled in the settings, the hardware version will not be detected the next time.
In the program, by default, R1 hardware program is programmed to CPLD for the first time and R2 hardware program is programmed to CPLD for the second time

In addition, according to my tests, I used the flag bit to be in a fixed state when I re-power on the persistent memory, so I used this bit state to determine if the device is first powered on and set the position to 0 after I power on

[image]

zhang00963, 2021-04-26:
@eried 
But I still feel “ui_ config” has a default value, because every time the persistent memory power is reconnected, “ui_ config” The values of are specific.

GullCode, 2021-04-26:
@zhang00963  it has a default value, but it is not consistent between H1 / H2 at least (I validated it myself since I have one of each, I even dumped the memory contents => different).
And maybe it is not even consistent for all H1 / H2 variant.
One can make some code based on the asset that bit X will be 0 on H1 and 1 on H2, but it has to be tested against all H1 and H2 variants. 
And it'll be a ugly piece of code X-D

zhang00963, 2021-04-26:
@GullCode 

Can you try [https://github.com/zhang00963/portapack-mayhem/tree/test](https://github.com/zhang00963/portapack-mayhem/tree/test)  this branch with your hardware

GullCode, 2021-04-26:
I will, in a few hours.

GullCode, 2021-04-26:
When I'm flashing your version on both my H1 and H2 they are booting OK. Note that they were already booting OK with the latest 1.4 (even with the latest next + my additions for my own app dev)

GullCode, 2021-04-26:
I experienced strange thing thought. I was testing 1.3.1,  eried next and yours. I had a couple of black screen while going from one version to another (of the firmware).
Solution was to power off entirely, change usb port (don't ask me why it helped), and reinstall via DFU (make program in a build directory).

zhang00963, 2021-04-26:
@GullCode 
Thank you for trying. What hardware versions of your H1 and H2 are R1 or R2?

GullCode, 2021-04-26:
H1 AK4951
H2 WM8731

zhang00963, 2021-04-26:
@GullCode  
The CPLD program version displayed in the interface?

GullCode, 2021-04-26:
Both of them are booting with 1.4.0
Information is coming from debug menu display under 1.4.0.
I've tested that sound is working on both.

zhang00963, 2021-04-26:
@GullCode 
The version is in the setup menu
As shown in the figure below:
[image]

GullCode, 2021-04-26:
Well. I don't have a working H1 after going from 1.4.0 to yours.

I tested with the wrong branch.

Your test version produce a blank screen on H1 that isn't disappearing.

zhang00963, 2021-04-27:
@GullCode 
Thank you for trying.
If your device works properly on v1.4.0, it means that your device is based on R2 revision.
My code branch, the first time to start is to program the R1 revised program. If you encounter a white screen or a black screen, you only need to restart the device to try again. If it works normally, you need to select the hardware version currently displayed in the settings, otherwise it will continue the process of hardware detection. The whole CPLD programming needs to last 5 ~ 10s, and you need to wait patiently.

If the restart still fails to make the device work normally, you need to separate the portapack and hackrf, wait a moment, and then assemble them together.

GullCode, 2021-04-27:
"...you only need to restart the device to try again. If it works normally,..." => My H1 does not have an internal RTCC battery yet, which means it's not going to save anything across reboots. It explains why it was keeping doing the finally freezing white screen test. In your testing system if you're using some persistent settings to detect a failed init or not, my H1 keeps playing the same detection which is obviously not working. And for the records when I tested I tried to wait at least 30 secs after reset and 30 more secs after full power off/on to see if an init was playing. 
I'm going to put a battery inside during the day to see if it's helping passing the test.
"If the restart still fails to make the device work normally, you need to separate the portapack and hackrf, wait a moment, then assemble them together and try again." => I never had the need to separate them. Usually booting the hackrf in DFU mode and reflashing it from here is enough.
Let's wait for some more free time during the day, I'll report back the tests when done :-)

zhang00963, 2021-04-27:
@GullCode  @eried 
are,If there is no RTC battery, your H2 is equipped with RTC battery, right?
In this case, the version information should be written into flash, so that it will not be affected by RTC battery.
I'll test later without RTC batteries.

eried, 2021-04-27:
Oh yeah, mine is with an RTC :/

GullCode, 2021-04-27:
My H1 have no RTCC but I can put one inside.
My H2 have one.

zhang00963, 2021-04-27:
@GullCode 
Er, may I have a picture of your H1?

GullCode, 2021-04-27:
Will do in the evening or tomorrow  after work

GullCode, 2021-04-28:
[image]

GullCode, 2021-04-28:
[image]
This is my H1

ZeroChaos-, 2021-04-30:
My green H1 fails when upgrading to 1.4.0 (works on 1.3.1).  I bisected to this commit by @zhang00963 

```
a0eaa70ff4720386e3c57960516221bfcc1d3443 is the first bad commit
commit a0eaa70ff4720386e3c57960516221bfcc1d3443
Author: zhang00963 <14372128+zhang00963@users.noreply.github.com>
Date:   Sun Apr 11 02:19:31 2021 +0800

    Realize the automatic recognition of audio chip, including ak4951en/wm8731/wm8731s,Try to fix the max2837 temperature problem

 firmware/application/apps/ui_debug.cpp |  3 ++-
 firmware/application/portapack.cpp     | 17 ++++++++++-------
 2 files changed, 12 insertions(+), 8 deletions(-)
```

I reverted the bit of code that disables model detection of R1 like so:
```
diff --git a/firmware/application/portapack.cpp b/firmware/application/portapack.cpp
index 44fc1db7..59cc682e 100644
--- a/firmware/application/portapack.cpp
+++ b/firmware/application/portapack.cpp
@@ -181,11 +181,11 @@ static PortaPackModel portapack_model() {
        if( !model.is_valid() ) {
                /*For the time being, it is impossible to distinguish the hardware of R1 and R2 from the software level*/
                /*At this point, I2c is not ready.*/
-               //if( audio_codec_wm8731.detected() ) {
-               //      model = PortaPackModel::R1_20150901;
-               //} else {
+               if( audio_codec_wm8731.detected() ) {
+                       model = PortaPackModel::R1_20150901;
+               } else {
                        model = PortaPackModel::R2_20170522;
-               //}
+               }
        }
 
        return model.value();
```
After that it works fine again.

sck-nogas, 2021-05-03:
> My green H1 fails when upgrading to 1.4.0 (works on 1.3.1). I bisected to this commit by @zhang00963
>
> ```
> a0eaa70ff4720386e3c57960516221bfcc1d3443 is the first bad commit
> commit [...]

I've done the same thing and that's what I'm running currently. Am waiting to see what pull requests @zhang00963 submits and gets accepted by @eried to fix this.

sck-nogas, 2021-05-28:
Just getting back from dealing with family issues. I pulled the latest code and re-compiled. Same issue still occurs. 

So reverted my changes back and...
```
diff --git a/firmware/application/portapack.cpp b/firmware/application/portapack.cpp
index 44fc1db7..59cc682e 100644
--- a/firmware/application/portapack.cpp
+++ b/firmware/application/portapack.cpp
@@ -181,11 +181,11 @@ static PortaPackModel portapack_model() {
        if( !model.is_valid() ) {
                /*For the time being, it is impossible to distinguish the hardware of R1 and R2 from the software level*/
                /*At this point, I2c is not ready.*/
-               //if( audio_codec_wm8731.detected() ) {
-               //      model = PortaPackModel::R1_20150901;
-               //} else {
+               if( audio_codec_wm8731.detected() ) {
+                       model = PortaPackModel::R1_20150901;
+               } else {
                        model = PortaPackModel::R2_20170522;
-               //}
+               }
        }
 
        return model.value();
```
This still works. So, what can I do to help test/improve the detection?

gigawatts, 2021-06-14:
Same issues here with an PortaPack H1 (20150901), v1.3.1 works fine, v1.4.0 just boots to a black screen. Thankfully its pretty easy to boot to DFU and flash back and forth between versions. 

Any progress on a fixed release, or anything I can provide to help?

ArjanOnwezen, 2021-06-14:
@zhang00963 Any progress on this issue?

zhang00963, 2021-06-15:
@gigawatts 
Can you try https://github.com/zhang00963/portapack-mayhem/tree/test this branch with your hardware，Can it work normally？

@ArjanOnwezen
Can you try this branch?( https://github.com/zhang00963/portapack-mayhem/tree/test),
However, because some devices are not equipped with RTC batteries, the detection mode of this branch will take effect. I am trying to record this information in flash

@sck-nogas
Yes, I didn't submit it because of a flaw in using the persistent memory method.
I'm trying to use flash to record the version information of the hardware, but I encountered a problem when writing. I can't write the data to flash normally. I don't know if there is a function that can be called to write the data.

gigawatts, 2021-06-16:
> @gigawatts
> Can you try https://github.com/zhang00963/portapack-mayhem/tree/test this branch with your hardware，Can it work normally？

Yep, I can give that a try. Do you have a compiled version of the equivalent `portapack-h1_h2-mayhem.bin` firmware from that branch? I have not previously tried compiling any portapack firmware on my own (I have not yet looked to see what the build environment looks like for doing so). Not a big deal if you don't, I can look up the build process (hopefully soon).

[62 more replies]

closed 2022-04-06 by jLynx