portapack-mayhem/mayhem-firmware #348: Microphone --->Tx

KB6mcc, 2021-04-25:
**Describe the bug**
CTCSS or any other tones and Roger beep are not working with the Microphone application.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Microphone and apply the CTCSS tone and enable roger beep'
2. Tap on 'TX'
3. Listen on other radio or with RF spectrum analyzer.

**Expected behavior**
No roger beep or ctcss tone detected on other receiver. 

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release V1.4.0
* Previous one V1.3.1 that work
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

eried, 2021-04-26:
I am not sure if the roger beep is included in the files on the SD card, but I guess they are not. Just to confirm, it was the same SD card for your test in 1.3.1, right?

KB6mcc, 2021-04-26:
I reformat the SD an updated with the new released V1.4.0 SD files before I flash with the V1.40 FW

deano817, 2021-04-29:
Same issue here on all 4 of my portapacks.

eried, 2021-04-30:
Thanks, I'll take a look over the weekend. Maybe @aldude999 has some insights, since he added few options?

aldude999, 2021-04-30:
The new code sends the audio_input buffer from proc_mictx.cpp to the dsp_modulate.cpp functions directly, which means the code in proc_mictx.cpp for the tone generation is skipped since they output to the variable "sample". It might be possible to run the beep generation code before executing the modulation, and change the modulate functions to accept a sample instead of a buffer. This would let you modify the audio input before it gets to the modulation side and insert tones. I wouldn't expect it to be much less efficient than the way it's done now.

backtopyramidone, 2021-05-17:
I just wanted to add that the ctcss implementation on soundboard is working as expected, not on mic though.

teixeluis, 2021-05-31:
I can also reproduce the absence of roger beep. I do notice however that the TX stays on a bit longer when roger beep is set.

ArjanOnwezen, 2022-03-30:
Duplicate of the more active #495

closed 2022-03-30 by ArjanOnwezen