portapack-mayhem/mayhem-firmware #349: PortaPack H2+ incompatible with my HackRF One?

edwardstorlien, 2021-04-29:
**Describe the bug**
I just got my new Portapack H2+ clone from AliExpress to use with my HackRF One clone. I have flashed the Mayhem firmware, and the HackRF boots just fine. The problem is that the display does not change when I try to scroll or click the buttons. It always shows the start menu. I have also tried the original Portapack firmware, but the problem is still the same.

I know that the HackRF works with the buttons and scroll wheel, because I can enter the receive mode and the Rx light will illuminate. But the display still shows the start menu. I can also scroll down (without actually seeing it scrolling down) to the option where I can use the HackRF with a computer over USB. When I enter that option, the display does turn blue and show the block diagram of the HackRF. It even shows the correct frequency and gain when I use software such as SDR Sharp.

I have tried the Portapack board with my friend's genuine HackRF One, and it works all fine. I also tried it with the original Portapack firmware. Therefore I think I can conclude that the problem lies in my HackRF, but the HackRF works all fine through a computer. Never experienced any problems, and it still works fine.

What could the problem be? Could there be some incompatibility between my HackRF One and the Portapack, since it works on my friend's?

**Affected versions**
- My HackRF One is a clone, and labeled 13 February 2014 on the PCB.
- I have tried the latest Mayhem firmware (ver. 1.4.0), 1.3.1 and 1.2.0
- I have also tried the original Portapack firmware

eried, 2021-04-30:
First, contact the seller and open a dispute if you are in time, it sounds like a hardware failure. If you touch the screen there is no reaction also?

edwardstorlien, 2021-04-30:
Could it be a hardware failure even though the Portapack worked on my friend's HackRF? Forgot to mention, but no. No reaction even if I touch the screen.

eried, 2021-04-30:
But your unit has never worked at all? why you have not complained to the seller?

edwardstorlien, 2021-05-01:
Sorry for the confusion...
My new PortaPack doesn't work on my HackRF, but it works on my friend's HackRF.
My HackRF works all fine without the PortaPack. Therefore I am unsure whether there is a problem with my HackRF or the PortaPack...

I have opened a dispute on the PortaPack now.

ArjanOnwezen, 2021-05-03:
@edwardstorlien Hopefully you're able to settle it with the seller.
@eried Since the portapack is working properly on another HackRF, this issue can be closed.

edwardstorlien, 2021-05-03:
Still remains a mystery, though... Closing the issue now!

closed 2021-05-03 by edwardstorlien

alebarra, 2022-03-07:
Same problem for me, i've HackRFOne with date 13 February 2014 on PCB, it works fine with portapack H1 (gridrf.com version) but not with H2.
does anyone have the hackrf one version dated 13 February 2014 with portpack H2 + working?

EricOwen1, 2022-07-28:
> Same problem for me, i've HackRFOne with date 13 February 2014 on PCB, it works fine with portapack H1 (gridrf.com version) but not with H2. does anyone have the hackrf one version dated 13 February 2014 with portpack H2 + working?

I also have the Hackrf One with portapack H2+. I don't know the date of the PCB. The next time I open it I will edit this and include that information. 
I had a similar problem.
I found that in order to enable the USB data connectivity and HackRF functionality with the Mayhem firmware I had to enable Hackrf mode. Otherwise the USB led does not light up.  I will have to search for information to find out exactly what HackRF mode does and how to manipulate the radio with it.