portapack-mayhem/mayhem-firmware #350: problem with generate_world_map.bin

Europe-Connection, 2021-05-01:
Hello I am trying to use generate_world_map.bin from Windows.
I open the file from cmd in the Python environment and I have an error:
 Line 26, in <module> 
outfile = open('../../sdcard/ADSB/world_map.bin', 'wb')
FileNotFoundError: Errno 2 No such file on directory '../../sdcard/ADSB/world_map.bin'

Can someone help me? Thank you

eried, 2021-05-02:
You can use the file that is included in the SD package, no need to recreate the map manually unless you want something specific for you.

Europe-Connection, 2021-05-02:
Thank you for your reply. In the package I have a world_map.jpg file. But the portapack puts me "no world_map.bin file in ADSB / directory" I do not have a .bin file in the directory to download on GitHub and I cannot generate it because of the error code mentioned above . I have put the ADSB file on my SD card and my SD card is functional because I can take captures and replay.

eried, 2021-05-03:
In the same section you downloaded the firmware, there is another file called `mayhem_vX.Y.Z_COPY_TO_SDCARD.7z`, that one contains everything. Do you have those files in the root of your sd card?

Europe-Connection, 2021-05-05:
Please excuse me for my ignorance. But I can't find the mayhem_vX.Y.Z_COPY_TO_SDCARD.7z file
I only found mayhem_v1.4.0_COPY_TO_SDCARD.7z from the page https://github.com/eried/portapack-mayhem/releases/tag/v1.4.0
Thanks a lot for your help.

eried, 2021-05-05:
Yes, that is the right file.

closed 2021-05-05 by eried

docjoens82, 2022-06-11:
i was able to generate the map you have to open python script and edit it, first off forward slash if for mac and backward slash is for windows when grabing file paths... ...FileNotFoundError: Errno 2 No such file on directory '../../sdcard/ADSB/world_map.bin'  need to edit the two spots in script that say ../../sdcard/ADSB/world_map.bin  for instance i put the folder  in c:\ drive level , so i edit script to say: outfile = open('C:\portapack-mayhem-next\sdcard\ADSB\world_map.bin', 'wb'), and also the line fix the im= line to this:
im = Image.open("C:\portapack-mayhem-next\sdcard\ADSB\world_map.jpg")  then run it and it will work.. its just a broken path and wrong forward slash vs backslashes ..

QIam1, 2023-04-08:
Is the complete inst. on how to do the Map bin setup...sorry fell like I need help.

gullradriel, 2023-04-08:
https://github.com/eried/portapack-mayhem/wiki/World-map-generation
git clone the depot to have the mentioned files

gullradriel, 2023-04-08:
While you're at it, feel free to explore the whole wiki !
https://github.com/eried/portapack-mayhem/wiki

eried, 2023-05-10:
Step by step for the map? or radio spectrum?

check the wiki for the map:
https://github.com/eried/portapack-mayhem/wiki/World-map-generation
https://github.com/eried/portapack-mayhem/wiki/Create-a-Custom-Map-with-Offline-Map-Maker-!

or just browse the index for something else:
https://github.com/eried/portapack-mayhem/wiki/