portapack-mayhem/mayhem-firmware #352: Jammer App don't work in v1.4.0

lsnakazone, 2021-05-09:
**Describe the bug**
Jammer App don't work in version 1.4.0 with Portapack H2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Transmit'
2. Tap on 'Jammer'
3. Mark 'Enable range'
4. Choose the start and stop frequency
5. Start the jammer

**Expected behavior**
No jammer signal was detected at any frequency used. 
Tests performed with Baofeng UV-5R and HackRF One + Portapack H2.

Obs.: The signal generator in the 'Tools option' works normally.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release: v1.4.0
* Previous one (if any) that did not presented the issue: v1.3.0
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

eried, 2021-05-09:
Thanks for the report, I will take a look with my cheap SDR

KB6mcc, 2021-05-10:
Jammer works with me with V1.4.0.......I only able to see it up to 2Ghz with my equipment.

lsnakazone, 2021-05-11:
I reinstalled the firmware and now it's working!

closed 2021-05-11 by lsnakazone