portapack-mayhem/mayhem-firmware #362: ADS-B data seems to be all over the place, regardless of aircraft signal quality and distance

teixeluis, 2021-05-18:
**Describe the bug**
When using the ADS-B app for listening for the aircraft beacons, it seems that half or 2/3 of the messages are not correctly decoded, resulting in incorrect position, altitude and speed data jumping around.

I live a few km from an international airport, and under the approach corridor, so I  can compare and have a clear view of the aircrafts and what the position should be.

**To Reproduce**

Open the ADS-B reception app, and select an aircraft with GPS data.

**Expected behavior**

Consistent information for every received beacon. Ability to ignore those with invalid checksum (?)

**Affected versions**

* Latest Stable Release

eried, 2021-05-18:
Can you provide some examples?

teixeluis, 2021-05-18:
> Can you provide some examples?

Air traffic is a bit calm right now, but I will add a video as soon as it picks up. It is very easy to reproduce as it is occuring with pretty much every aircraft I look at.

Cheers

teixeluis, 2021-05-18:
@eried here is the video reproducing the behavior:

https://photos.app.goo.gl/zhDWqZx4M4UTBP3aA

eried, 2021-05-19:
> It's like the airliner has a warp drive on it.

hahaha yeah, thanks the video is really helpful. It is certainly some issue with a conversion and decimal places

euquiq, 2021-05-22:
This was working fine before last PRs applied. It may also be related to some error checking not being done ? in any case, maybe @jwetzell can better understand what is going on ? (he authored the latests Prs involving adsb).

jwetzell, 2021-05-22:
ADSB logs packet data (before decoding) to SD card I believe (there might be an option for it). But that might help in seeing where decoding is going wrong.

teixeluis, 2021-05-22:
@jwetzell  I have this for now. Not a lot of data, but is what I could get so far, as air traffic is calm lately (different story before the pandemic). If you look at ICAO:39CEA2 you will see a lot of incorrect coordinates. It goes from:

20210520194015 8D49528D6019C56379ADBDB321A9 ICAO:49528D Alt:4100 Lat:38.28 Lon:-9.85

to

20210520194033 8D49528D6019056041ACF58C6268 ICAO:49528D Alt:3800 Lat:38.09 Lon:-9.97

and then to:

20210520194053 8D49528D6017955CF1AC24179C0E ICAO:49528D Alt:3625 Lat:38.89 Lon:-9.10

within the same minute. These two last coordinates are about 90 km apart! Doing the math, that is about 16200 Km/h.
For an object flying at 3625 ft, it would be something else :smile: 

Cheers

teixeluis, 2021-05-25:
Here is an updated file with more examples. It seems to have a nearly regular alternating pattern.

[adsb.txt](https://github.com/eried/portapack-mayhem/files/6541647/adsb.txt)

teixeluis, 2021-06-11:
hi @jwetzell do you have any clue on what could be going on here?

jwetzell, 2021-06-11:
I haven't had much time to look into it my initial guess would be some questionable rounding errors in the calculation for Lat/Lon from the Odd/Even frames. I know a few bits of the math are kind of estimated for performance reasons.

teixeluis, 2021-06-18:
> I haven't had much time to look into it my initial guess would be some questionable rounding errors in the calculation for Lat/Lon from the Odd/Even frames. I know a few bits of the math are kind of estimated for performance reasons.

Ok makes sense. On the other hand if we look at some of the examples, there are also some implausible altitude jumps as well. And altitude is supposedly a more straightforward calculation. I will take a closer look later (i.e. manually calculate the correct coordinates from the frames) from the raw frames to see what we get.

teixeluis, 2021-06-19:
> I know a few bits of the math are kind of estimated for performance reasons.

@jwetzell you mean in particular for the longitude calculations, where instead of the doing the math for:

[image]

we are instead comparing with some pre-calculated values from a lookup table?

```
int cpr_NL(float lat) {
	if (lat < 0)
		lat = -lat;		// Symmetry
	
	for (size_t c = 0; c < 58; c++) {
		if (lat < adsb_lat_lut[c])
			return 59 - c;
	}
	
	return 1;
}
```

[adsb.cpp#L149](https://github.com/eried/portapack-mayhem/blob/8afd83a4e18fd60312e4641912c8ebd6453b5b7a/firmware/common/adsb.cpp#L149)

The rest of the code seems to be performing regular floating point calculations, so errors should be negligible..

While looking at how odd and even frames are being correlated, I see that there is a 20 second timeout between the two frames:

[ui_adsb_rx.hpp#L85](https://github.com/eried/portapack-mayhem/blob/8afd83a4e18fd60312e4641912c8ebd6453b5b7a/firmware/application
/apps/ui_adsb_rx.hpp#L85)

I couldn't find any official info on what should be the max value, but would challenge if the value is too high?

Thanks

Cheers

jwetzell, 2021-06-19:
Yeah, things like that. I just plugged the packets you gave into a proper ADSB packet decoder and got just slightly different decimal parts on the lat/lon then what was output in the debug making me thing about some shotty calculations. That timeout is interesting. This is the side of the ADSB code I never really looked at as it already existed and I don't really get any air traffic to be able to reliably collect any sort of data.

teixeluis, 2021-06-19:
Ok @jwetzell, thanks for the feedback. Did you try calculating the NL with the cmath functions? If so, the performance penalty was noticeable?

jwetzell, 2021-06-19:
I have not tried to change these calculations in anyway like I side because this is code that already existed before I started to work on the ADSB stuff (mainly adding the heading and speed decoding) which is a completely different packet from the position data

teixeluis, 2021-06-19:
Tried replacing the NL function with the complete math calculations:

```
int cpr_NL_precise(float lat) {
	return (int) floor(2 * PI / acos(1 - ((1 - cos(PI / (2 * NZ))) / pow(cos(PI * lat / 180), 2))));
}
```
The behavior seems to be the same. Didn't find any noticeable performance degradation though.

However after looking closer at the jump in the position, I could see that it is coincident to that of another aircraft in that position. It seems to alternate between one and another, while still associated to the same ICAO address.

This suggests that there can be a bug elsewhere, but the decoding being correct..

I am sharing the log with this test from today.

[adsb.txt](https://github.com/eried/portapack-mayhem/files/6681405/adsb.txt)

teixeluis, 2021-06-19:
btw created this branch for those wanting to experiment with the alternate NL function:

https://github.com/teixeluis/portapack-mayhem/tree/hotfix-adsb-position

teixeluis, 2021-06-23:
@jwetzell, I got a couple more findings on this topic:

 * one misleading aspect was the rendering of the latitude and longitude in the ads-b app detail screen, and also in the log output - both use the same conversion which is not implemented correctly:
 
https://github.com/eried/portapack-mayhem/blob/next/firmware/application/apps/ui_adsb_rx.cpp#L235

 * after looking more carefully at the pairs of frames when the jumps occur, I could clearly see that the selected aircraft remains the same, but the frames that appear are not always from the same ICAO address. Also the calculated coordinates will change, when odd and even frames containing position data from a different aircraft are decoded while we are looking at the details from the first aircraft. At a first glance this seems to be UI related..

Cheers

teixeluis, 2021-06-23:
That was it:

Here:

https://github.com/eried/portapack-mayhem/blob/next/firmware/application/apps/ui_adsb_rx.cpp#L242

and here:

https://github.com/eried/portapack-mayhem/blob/next/firmware/application/apps/ui_adsb_rx.cpp#L250

We should be checking if the the ICAO address from the frame we are receiving matches the address from our current entry in the details view.

Created a fix for this which is available in this branch:

https://github.com/teixeluis/portapack-mayhem/tree/hotfix-adsb-position

Will do a second review and check if improvements are necessary and submit a PR afterwards.

Feel free to test the branch nevertheless.

Enjoy

Cheers

ArjanOnwezen, 2021-11-09:
Hi @teixeluis, I don't experience this issue (anymore). 
It might be solved with your fix or with #422, which is also loaded on my portapack.
Could you please check, so this issue might be closed.

ArjanOnwezen, 2021-11-24:
Fixed in new release, probably due to #422

closed 2021-11-24 by ArjanOnwezen