portapack-mayhem/mayhem-firmware #363: 1.40 firmware brakes hackrf_sweep

0181505050, 2021-05-18:
https://github.com/pavsa/hackrf-spectrum-analyzer 

does not work with latest firmware in Linux or in Windows.

ArjanOnwezen, 2021-05-18:
This is probably related to  updated HackRF submodule to mossmann/hackrf@5f1536c

0181505050, 2021-05-18:
Inside Hack RF Sweep you can change the frequency and it shows it changing on the device. but there is no readout at all. and says HackRF is disconnected.

ArjanOnwezen, 2021-05-18:
Did you already updated your tools to the latest version? https://github.com/pavsa/hackrf-spectrum-analyzer/issues/38

0181505050, 2021-05-19:
> Did you already update libhackrf and hackrf-tools on your host computer. [pavsa/hackrf-spectrum-analyzer#38](https://github.com/pavsa/hackrf-spectrum-analyzer/issues/38)

Sorry, i am still learning Linux, where can i download the updated versions? when i go to the package manager there are no updates for it.

0181505050, 2021-05-19:
Would you mind telling me how this is done on manjaro? I cant find libhackrf and hackrf-tools in the add remove package manager. sorry i am still learning.

ArjanOnwezen, 2021-05-19:
Hi @0181505050  I'm also new to the project, bit was just citing the hackrf-spectrum-analyzer project site.

0181505050, 2021-05-19:
> Hi @0181505050 I'm also new to the project, bit was just citing the hackrf-spectrum-analyzer project site.

ok no problem. I can't believe I am the only one. Does it work for you?

lm410700, 2021-05-27:
I'm just getting into this HackRF and hackRF_sweep Spectrum Analyzer interface (I'm using Windows 10) was one of my primary tests. I get working software with Mayhem version 1.3.1 and broken with 1.4.0. Behavior is consistent as I flash back and forth. Blue Portapack screen reflects start frequency, but no spectrum feedback to Win 10 sweep software. **I also believe it is associated with the HackRF firmware base used under Mayhem**, as when I flash just the HackRF FW I get the same results between the 2021 version and the 2018 version... working versus broken sweep viewing in Windows 10. So, for me the last working version is either Mayhem 1.3.1 with blue screen feedback on Portapack, or "raw" HackRF 2018 with no Portapack feedback. BTW, I'm using the Mayhem flash_hackrf_one.bat tool and swapping various hackrf_one_usb.bin versions to flash .bins for HackRF "raw" firmware testing.

0181505050, 2021-05-31:
Glad someone else has this issue and its not just me. Hopefully it will be looked at for the next version,

ArjanOnwezen, 2021-06-10:
This needs to be fixed by the author (pavsa) or someone else in the hackrf-spectrum-analyzer, since it's not (yet) compatible with the latest version of hackrf firmware.
https://github.com/pavsa/hackrf-spectrum-analyzer/issues/38

closed 2021-06-10 by ArjanOnwezen