portapack-mayhem/mayhem-firmware #365: Microphone app crashes when enabling RX listening before or after TX in Portapack H2 w/WM8731 audio chip

teixeluis, 2021-05-20:
**Describe the bug**

The device will hang when RX audio listening is enabled before or after transmitting with the app.

**To Reproduce**

Steps to reproduce the behavior:

1. Go to Microphone TX app
2. Tap TX 
3. Check "RX audio listening"

or

2. Check "RX audio listening"
3. Tap TX

**Expected behavior**

Normal PTT operation with audio listening after transmitting.

**Affected versions**

1.4.0

**Additional**

I have also added screenshots and details in the related pull request:

https://github.com/eried/portapack-mayhem/pull/329

Thanks

Cheers

ArjanOnwezen, 2021-05-21:
I guess it's indeed a WM8731 specific issue, since it does work on the AK4951 (at least on the one I have).

teixeluis, 2021-05-21:
I have further found that it is not just with the microphone app, but with any other app that does TX: if after using it I try to use a RX app, the device hangs and the RX LED starts blinking.

9H1CG, 2021-05-21:
I have the same issue, but I do not even need to Check "RX audio listening". Going back to Receive from any TX mode crashes the Portapack, the RX led keeps blinking and I have to reset.

teixeluis, 2021-05-22:
As I have commented in discord, after some digging and giving more attention to previous findings around TX/RX switching, I first found that the device was hanging because it enters this halt command:

[](https://github.com/eried/portapack-mayhem/blob/433529b4cb80fadabd918527283055cc97ce2fc3/firmware/application/radio.cpp#L119)

Other users found that this load_sram code would render the TX/RX switching too slow for their application, but in this case (eventually due to some hardware difference), load_sram fails, and the halt is deriberately done in this situation.

As mentioned there, the CPLD code had improvements (in the main hackrf repo) that apparently make the switching between sram and eeprom code unnecessary when switching between TX and RX.

As such, by removing the code that performs hackrf::cpld::load_sram():

`
void set_direction(const rf::Direction new_direction) {
	/* TODO: Refactor all the various "Direction" enumerations into one. */
	/* TODO: Only make changes if direction changes, but beware of clock enabling. */
	
	// Hack to fix the CPLD (clocking ?) bug: toggle CPLD SRAM overlay depending on new direction
	// Use CPLD's EEPROM config when transmitting
	// Use the SRAM overlay when receiving
	if (direction != new_direction) {
		if (new_direction == rf::Direction::Transmit) {
			hackrf::cpld::init_from_eeprom();
		} /* else {
			// TODO teixeluis: analyse better why we might be entering here (issue #365):
			if( !hackrf::cpld::load_sram() ) {
				chSysHalt();
			}
		} */
	}
	
	direction = new_direction;
	
	second_if.set_mode((direction == rf::Direction::Transmit) ? max2837::Mode::Transmit : max2837::Mode::Receive);
	rf_path.set_direction(direction);

	baseband_codec.set_mode((direction == rf::Direction::Transmit) ? max5864::Mode::Transmit : max5864::Mode::Receive);

	if (direction == rf::Direction::Receive)
		led_rx.on();
	else
		led_tx.on();
}
 `
And by updating the CPLD manually:

`
git clone https://github.com/mossmann/hackrf.git

hackrf_cpldjtag -x hackrf/firmware/cpld/sgpio_if/default.xsvf
`
It sucessfully fixed this problem. 

I performed a few tests with some of the applications, including microphone, and so far I had no issues. Will open a PR with this change.

I guess users will have to update the CPLD as well, after this change. I have first tested the change without the CPLD update and the RX audio seemed distorted.

ArjanOnwezen, 2021-05-26:
I'll try to load the firmware on my PortaPack tonight (I have a Chinese clone with AK chip)

ArjanOnwezen, 2021-05-26:
Tested the commit on my Portapack with AK chip. Switching between TX and RX works fine, like it does on the current (1.4.0) firmware.

teixeluis, 2021-06-24:
Closing the issue, since it is already addressed and merged to next.

closed 2021-06-24 by teixeluis