portapack-mayhem/mayhem-firmware #368: Microphone app - AM/SSB/DSB issues

teixeluis, 2021-05-22:
**Describe the bug**

Trying these new modes, I had the following behavior:

- In AM the carrier signal is very weak compared to NFM. This mode seems more like DSB;
- In any of the modes, modulation is pretty weak, regardless of the amplifier settings or microphone gain;

**To Reproduce**
Steps to reproduce the behavior:

Test the microphone app in AM/SSB/DSB. Check the transmitted signal with a spectrum analyzer on another SDR.

**Expected behavior**

Same carrier signal power as in NFM when in AM - for example HDSDR cannot properly demodulate the signal as AM.
More signal output in any of these new modes.

**Affected versions**

Version 1.4.0

**Additional**

Added more evidences here (including .wav baseband file with tests performed in all of the modes):

[](https://github.com/eried/portapack-mayhem/pull/329#issuecomment-841829259)

teixeluis, 2021-06-24:
This issue is reproducible in devices with the WM8731 chip. Anyone with the AK4951 audio chip confirms the same issue?

It is helpful to understand the potential differences, in order to help isolate the cause.

@aldude999 in what devices have you tested the feature when you developed it?

Thanks

Cheers

jumbo77, 2021-08-08:
i have ak4951 version ,i test it is ok,the bug just for wm8731 version.

Brumi-2021, 2022-01-01:
Hi, I think that this open issue #368 , 
(Microphone app - AM/SSB/DSB issues #368 )
Now , it could be closed , once accept and merge that  PR https://github.com/eried/portapack-mayhem/pull/457

ArjanOnwezen, 2022-01-13:
Implemented in version 1.4.3

closed 2022-01-13 by ArjanOnwezen