portapack-mayhem/mayhem-firmware #370: Radiosondes app - add QR code generator for GPS coordinates of radiosonde.

teixeluis, 2021-05-30:
**Describe the solution you'd like**
Given that the embedded maps feature cannot be expected to in any way be equivalent to an online maps service, in order to maximize how useful this app can be in the field, I propose the addition of a QR code generator that can encode the GPS position of the radiosonde into a Google maps URL or equivalent. This way a user can quickly scan the current position and use it during the chase.

**Describe alternatives you've considered**
An alternative, albeit more limited approach, could be to present the coordinates in the form of a text URL which some smartphones can OCR scan.

ArjanOnwezen, 2022-01-03:
Implemented with #460

ArjanOnwezen, 2022-01-13:
Implemented in version 1.4.3