portapack-mayhem/mayhem-firmware #371: Radiosondes app - add beeping sound when message is received

teixeluis, 2021-05-31:
**Describe the solution you'd like**

During the final stage of the search, it may be difficult to spot the sonde, as it can be hidden between dense vegetation. Analyzing the direction of the signal, and its strength is particularly useful in spotting the exact location of the device.
With the brightness of daylight it can however be difficult to look at the screen to properly see the rssi bars.

By playing an audible tone in the speaker (or headphones) that varies in frequency depending on the RSSI everytime a message from the sonde is received, I think is a feature that would greatly improve the app in field use.

teixeluis, 2021-06-08:
Implemented feature. It can be found here:

https://github.com/eried/portapack-mayhem/pull/374

RSSI based pitch is still work in progress.

eried, 2021-06-09:
Thanks a lot @teixeluis

closed 2021-06-09 by eried

teixeluis, 2021-06-09:
Already getting the RSSI events. Still need to improve the tone pitch range. Will submit new PR once it is nice.